Zendure/Zendure-HA #522: Freeze still happens with pre13

pulsarz, 2025-08-02:
So just noticed one of my 2 hypers was frozen.
1 was charging while the other was discharging.
I use local mqtt and use EMQX as server.

Turned off the hyper and back on and it works again.

[image]

[image]

pulsarz, 2025-08-03:
Master firmware v2.1.30 btw

pulsarz, 2025-08-05:
Had the same thing happen again.
Restarting EMQX resolved the issue (usually does).

Even tried giving a static IP and switching back to 2.4Ghz but it has not helped.

FireSon, 2025-08-05:
That is interesting, is there any logging in EMQX regarding the connection with your hyper? If you are 'only' resetting EMQX, is it not likely that it has nothing to do with the Hyper, but more with EMQX?

pulsarz, 2025-08-10:
Still happening on 1.1.0.
Have adjusted various options in the broker but no dice.

And it keeps happening to the same hyper. The 3.0 revision hyper has had zero freezes since I got it 2 weeks ago..

closed 2025-09-09 by FireSon