Zendure/Zendure-HA #527: Benötigt man noch die Intelligente Steuerung in der Zendure APP?

ralle6677, 2025-08-02:
Hallo,

in Verbindung mit einem Hyper stellt sich mir folgende Frage: Welche Einstellung sollte man in der Zendure App nutzen, wenn man die die HA-Integration verwendet?

Benötigt man in der Zendure App / Energieplan noch den Intelligenten Modus? 

DANKE!!

CommanderROR9, 2025-08-02:
Ich schließe mich der Frage mal an...ich finde ws grundsätzlich OK, wenn das Zendure HEMS mit den Ecotracker die grundsätzliche Regelung übernimmt. Aber ich will z.B. den Max SoC über sen Tah verteilt steuern, und im Winter zu bestimmten Zeiten aus dem Netz laden (via HA Automatisierung). 
Ist da dann trotzdem "alles in HA" der richtige Weg?

harrymayr, 2025-08-02:
> Ich schließe mich der Frage mal an...ich finde ws grundsätzlich OK, wenn das Zendure HEMS mit den Ecotracker die grundsätzliche Regelung übernimmt. Aber ich will z.B. den Max SoC über sen Tah [...]

Prinzipiell sind die Geräte selbst erst mal doof. Die bekommen gesagt, wieviel Leistung sie in oder aus der Batterie ins Netz schieben sollen. Dazu muss **einer** den Hut aufhaben. Es geht nicht, dass HEMS oder die Integration gleichzeitig laufen, da es nie die gleichen Kommandos gibt. HEMS läuft auf irgendeinem Server und schickt per MQTT / HTTP den Geräten die Anweisungen und die Integration macht das genauso. Beides zusammen geht nicht.
Und ich wüsste auch nicht, was HEMS kann, was die Integration nicht kann. Die Integration hat meiner Meinung nach mehr Möglichkeiten, so ermöglicht sie mir, mit einem nicht App-kompatiblen Sensor für meinen Stromverbrauch, den Hyper zu steuern. Ich könnte z.B. gar nicht HEMS nutzen, weil mein "Volkszähler" nicht kompatibel ist.

Sepp2812, 2025-08-03:
> > Benötigt man in der Zendure App / Energieplan noch den Intelligenten Modus?
> 
> Wenn du den ZendureManager mit der Smarten Leistungsregelung verwenden willst, solltest du alle Energiepläne stoppen. Auch HEMS solltest du nicht verwenden. Um mal ein Film-Zitat zu verwenden: "Es kann nur Einen geben" 😅

Servus harrymayr

ich möchte alles über den ZendureManger (Integration steuern) was muss ich da alles einstellen ohne Appeinstellungen?
Habe einen Hyper 2000 , den ich nur als AC Speicher hernehme (also keine PV Module angeschlossen, habe eine PV Anlage)
Shelly Pro 3EM

Lokalen Mosquitto MQTT AddOn verwenden--funktioniert irgend wie nicht
unter P1 Sensor steht sensor.power_actual drinn, habe aber es schon mit den sensor.shellypro3em_34987a457c08_total_active_power
probiert

was ich bis jetz probiert habe ist
AC Betriebsmodus -- AC Eingangsmodus---Einganslimit auf 1200 Watt
Betribsmodus ----- Smarte Leistungsregelung
Gerätecluster---Gerät hat einen eigenen Stromkreis

wenn ich unter Energieplan alles stoppe lädt er zwar mit 1200 Watt aus dem Netz sollte er aber nicht, nur bei Überschussverfolgung, ausserdem besteht dann keine Verbindung zum Shelly nicht mehr (Überwachungsschaltung)

heist also, sobald ich in der App unter Energieplan -Intelligent das "Ausführen" stoppe ist Schluß mit Überschussstromverfolgung und Shellyanbindung

darum wär es gut, wenn man das Hyper Energy Programm auch steuern könnte https://github.com/Zendure/Zendure-HA/issues/528
oder hab ich hier irgendwo einen Gedankenfehler.

Gruß
Sepp

harrymayr, 2025-08-03:
Die pre13 (neueste Version ist pre14) läuft aber bei euch im Cloud Mode?

Zu den Einstellungen (so funktioniert es bei mir, auch einen Hyper mit 3 Batterien nur AC Laden/Entladen):

- Beta API und Token leer 
- Zendure Benutzer-Name und PW
- P1 Sensor habe ich verschiedene Sachen getestet, die alle funktioniert haben. Ich konnte diese sowohl bei der einfachen Konfiguration, sowie bei Neu Konfigurieren angeben, und die wurden auch bei einem neuen Aufruf der entsprechenden Konfiguration angezeigt. 
  - einen template Sensor, mit Namen power_actual 
  - einen template Sensor, mit Namen shellypro3em_34987a457c08_total_active_power
  - meinen Sensor vom "Volkszähler"
- MQTT log und local MQTT angehakt
nächste Seite:
- die IP-Adresse meiner HA Installation (ich nutze den Standard MQTT Broker)
- den mqttuser und das PW was ich auch benutze für den MQTT Explorer
- meine WiFi SSID und das PW

Bei der Änderung von Zeundure, MQTT oder WiFi Daten würde ich HA immer neu starten.

Weiterhin habe ich die Erfahrung gemacht, dass bei der Nutzung der Zendure-Manager Smart-Modes das Ausgangs- und Eingangs-Limit auf 0 stehen sollte. Das habe ich mit einer Automation gelöst.
In der App kein HEMS und kein Energie-Programm aktiviert.
Für LocalMQTT wird zwingend eine BT-Verbindung benötigt, der Hyper muss also in der BT Advertisement-Liste auftauchen.
Manchmal ist es bei mir nötig gewesen, den iot-Knopf des Hyper für ca. 5-6 Sekunden zu drücken. Der fährt dann bei mir nicht runter, sondern fängt an die WiFi-Verbindung (und BT) neu aufzubauen.
Was auch noch hilft ist, immer mal wieder die Neu Konfiguration aufzurufen und immer nur ohne weitere Änderungen mit OK zu bestätigen. 

Das sollte eigentlich funktionieren. Wenn nicht, dann bitte das debug-log aktivieren, HA neu starten, für ca. 5 Minuten warten und mir das log an harryzagithub@gmail.com zusenden. Dann schaue ich mal nach, ob ich das Problem sehe.

CommanderROR9, 2025-08-03:
Ich hatte in Kombination mit dem SF2400AC ja schonmal das Problem, dass der lokale Zugriff via BLE nicht geht. Der Faktor sollte aber dann ja mit der nächsten Version der Zendure Firmware gelöst sein.
Aber was das AC Laden/Entladen anging...da hatte ich garkeinen Erfolg mit der Version die ich getestet hatte.
Grundsätzlich wurde der Sensor den ich in HA dafür konfiguriert hatte angenommen und die Werte wurde auch weitergeleitet an die Integration, aber der SF 2400AC reagierte darauf... garnicht. Wenn ich die "Limits" auf 0 gestellt hatte hat sich garnichts bewegt, und wenn ich sie auf einen anderen Wert gestellt habe dann wurde entweder geladen oder entladen, aber nicht flexibel auf Input und Output reagiert.
Ich muss es wohl mal noch mit der aktuellen pre13 oder pre14 testen.
Ich denke ich warte aber erstmal auf die neue Firmware von Zendure.

CommanderROR9, 2025-08-03:
Also...ich habe es eben mal kurz getestet:
Version pre-14 im ZenSDK Modus. HEMS in der App deaktiviert und "Smarte Steuerung" in den Integration aktiviert. Der Ecotracker war als Power Sensor eingetragen und konfiguriert.

Das Ergebnis: Solange Input und Output Limit auf "0" stehen passiert nichts. Das System lädt nicht und entlädt nicht, egal ob Überschuss vorhanden oder nicht.
Nur wenn ich die beiden Regler für Input und Output Limit bewege, dann wird auch Beladen/Entladen, allerdings nur, wenn ich gleichzeitig jeweils den AC Modus von AC Laden auf AC Entladen umschalte.
Also momentan muss ich weiterhin die Smarte Regelung über die App verwenden und die HA Integration lediglich für das Einstellen des Max. SoC und für die Aufzeichnung der Daten verwenden.

Sepp2812, 2025-08-03:
> Die pre13 (neueste Version ist pre14) läuft aber bei euch im Cloud Mode?
> 
> Zu den Einstellungen (so funktioniert es bei mir, auch einen Hyper mit 3 Batterien nur AC Laden/Entladen):
> 
>     * Beta API und Token leer
> 
>     * Zendure Benutzer-Name und PW
> 
>     * P1 Sensor habe ich verschiedene Sachen getestet, die alle funktioniert haben. Ich konnte diese sowohl bei der einfachen Konfiguration, sowie bei Neu Konfigurieren angeben, und die wurden auch bei einem neuen Aufruf der entsprechenden Konfiguration angezeigt.
>       
>       * einen template Sensor, mit Namen power_actual
>       * einen template Sensor, mit Namen shellypro3em_34987a457c08_total_active_power
>       * meinen Sensor vom "Volkszähler"
> 
>     * MQTT log und local MQTT angehakt
> 
> 
> nächste Seite:
> 
>     * die IP-Adresse meiner HA Installation (ich nutze den Standard MQTT Broker)
> 
>     * den mqttuser und das PW was ich auch benutze für den MQTT Explorer
> 
>     * meine WiFi SSID und das PW
> 
> 
> Bei der Änderung von Zendure, MQTT oder WiFi Daten würde ich HA immer neu starten.
> 
> Weiterhin habe ich die Erfahrung gemacht, dass bei der Nutzung der Zendure-Manager Smart-Modes das Ausgangs- und Eingangs-Limit auf 0 stehen sollte. Das habe ich mit einer Automation gelöst. In der App kein HEMS und kein Energie-Programm aktiviert. Für LocalMQTT wird zwingend eine BT-Verbindung benötigt, der Hyper muss also in der BT Advertisement-Liste auftauchen. Manchmal ist es bei mir nötig gewesen, den iot-Knopf des Hyper für ca. 5-6 Sekunden zu drücken. Der fährt dann bei mir nicht runter, sondern fängt an die WiFi-Verbindung (und BT) neu aufzubauen. Was auch noch hilft ist, immer mal wieder die Neu Konfiguration aufzurufen und immer nur ohne weitere Änderungen mit OK zu bestätigen.
> 
> Das sollte eigentlich funktionieren. Wenn nicht, dann bitte das debug-log aktivieren, HA neu starten, für ca. 5 Minuten warten und mir das log an [harryzagithub@gmail.com](mailto:harryzagithub@gmail.com) zusenden. Dann schaue ich mal nach, ob ich das Problem sehe.

hm, der verlangt bei mir immer den Token (in der 13 und 14 er Version), wo bekomm ich den diesen her ?

wie schon hier nachgefragt https://github.com/Zendure/Zendure-HA/issues/528

das wär super wenn man das steuern könnte

ich glaube das ist alles zu viel für mich, auf jeden fall Danke schon mal für die Beschreibeung
wenn ich viel Zeit habe, werde ich es nochmal Versuchen

Gruß
Sepp

harrymayr, 2025-08-03:
Token: Siehe #534

Sepp2812, 2025-08-03:
Danke

CommanderROR9, 2025-08-03:
Ich probiere es nachher mal mit dem.Log, wenn ich am Rechner bin.

CommanderROR9, 2025-08-03:
2025-08-03 12:02:21.923 INFO (paho-mqtt-client-Q5GxX6Tw) [custom_components.zendure_ha.api] Topic: /BC8B7F/SolarFlow 2400 AC/properties/report => {'timestamp': 1754215340, 'messageId': 196233, 'version': 2, 'product': 'solarFlow2400AC', 'properties': {'packNum': 2}, 'packData': [{'sn': 'FO4NHN4N2000314'}, {'sn': 'FO4NHN4N2000468'}]}
2025-08-03 12:02:21.939 INFO (paho-mqtt-client-Q5GxX6Tw) [custom_components.zendure_ha.api] Topic: /BC8B7F/SolarFlow 2400 AC/properties/report => {'timestamp': 1754215340, 'messageId': 196232, 'version': 2, 'product': 'solarFlow2400AC', 'properties': {'heatState': 0, 'packInputPower': 0, 'outputPackPower': 0, 'outputHomePower': 0, 'remainOutTime': 14730, 'packState': 2, 'electricLevel': 74, 'gridInputPower': 107, 'solarInputPower': 0, 'solarPower1': 0, 'solarPower2': 0, 'solarPower3': 0, 'solarPower4': 0, 'solarPower5': 0, 'solarPower6': 0, 'pass': 3, 'reverseState': 0, 'socStatus': 0, 'hyperTmp': 3001, 'gridOffPower': 83, 'dcStatus': 1, 'pvStatus': 0, 'acStatus': 2, 'dataReady': 1, 'gridState': 1, 'BatVolt': 5002, 'socLimit': 0, 'writeRsp': 0, 'acMode': 1, 'inputLimit': 0, 'outputLimit': 0, 'socSet': 800, 'minSoc': 100, 'gridStandard': 0, 'gridReverse': 2, 'inverseMaxPower': 2000, 'lampSwitch': 1, 'gridOffMode': 0, 'IOTState': 2, 'fanSwitch': 1, 'fanSpeed': 0, 'bindstate': 0, 'VoltWakeup': 0, 'OldMode': 0, 'OTAState': 0, 'LCNState': 0, 'factoryModeState': 0, 'ts': 1754215336, 'tsZone': 14, 'smartMode': 0, 'chargeMaxLimit': 2400, 'phaseSwitch': 1, 'rssi': -72, 'aiState': 0}, 'packData': [{'sn': 'FO4NHN4N2000314', 'packType': 5, 'socLevel': 74, 'state': 2, 'power': 9, 'maxTemp': 3011, 'totalVol': 4990, 'batcur': 65534, 'maxVol': 333, 'minVol': 333, 'softVersion': 4101, 'heatState': 0}, {'sn': 'FO4NHN4N2000468', 'packType': 5, 'socLevel': 74, 'state': 2, 'power': 10, 'maxTemp': 2991, 'totalVol': 5000, 'batcur': 65534, 'maxVol': 333, 'minVol': 333, 'softVersion': 4101, 'heatState': 0}]}
2025-08-03 12:02:23.376 WARNING (MainThread) [homeassistant.components.mqtt.sensor] Ignoring invalid option received on topic 'ebusd/hmu/State', got '128', allowed: ready, heating, error, cooling, heating_water
2025-08-03 12:02:24.343 INFO (MainThread) [custom_components.zendure_ha.manager] Update p1: -1 power: 0.0 operation: 2 delta:-1.125 stddev: 3.3330729064933458 fast: False
2025-08-03 12:02:24.343 INFO (MainThread) [custom_components.zendure_ha.manager] Unable to charge/discharge p1: -1
2025-08-03 12:02:26.921 INFO (paho-mqtt-client-Q5GxX6Tw) [custom_components.zendure_ha.api] Topic: /BC8B7F/SolarFlow 2400 AC/properties/report => {'timestamp': 1754215345, 'messageId': 196235, 'version': 2, 'product': 'solarFlow2400AC', 'properties': {'packNum': 2}, 'packData': [{'sn': 'FO4NHN4N2000314'}, {'sn': 'FO4NHN4N2000468'}]}
2025-08-03 12:02:26.938 INFO (paho-mqtt-client-Q5GxX6Tw) [custom_components.zendure_ha.api] Topic: /BC8B7F/SolarFlow 2400 AC/properties/report => {'timestamp': 1754215345, 'messageId': 196234, 'version': 2, 'product': 'solarFlow2400AC', 'properties': {'heatState': 0, 'packInputPower': 0, 'outputPackPower': 0, 'outputHomePower': 0, 'remainOutTime': 14730, 'packState': 2, 'electricLevel': 74, 'gridInputPower': 114, 'solarInputPower': 0, 'solarPower1': 0, 'solarPower2': 0, 'solarPower3': 0, 'solarPower4': 0, 'solarPower5': 0, 'solarPower6': 0, 'pass': 3, 'reverseState': 0, 'socStatus': 0, 'hyperTmp': 3001, 'gridOffPower': 84, 'dcStatus': 1, 'pvStatus': 0, 'acStatus': 2, 'dataReady': 1, 'gridState': 1, 'BatVolt': 5002, 'socLimit': 0, 'writeRsp': 0, 'acMode': 1, 'inputLimit': 0, 'outputLimit': 0, 'socSet': 800, 'minSoc': 100, 'gridStandard': 0, 'gridReverse': 2, 'inverseMaxPower': 2000, 'lampSwitch': 1, 'gridOffMode': 0, 'IOTState': 2, 'fanSwitch': 1, 'fanSpeed': 0, 'bindstate': 0, 'VoltWakeup': 0, 'OldMode': 0, 'OTAState': 0, 'LCNState': 0, 'factoryModeState': 0, 'ts': 1754215341, 'tsZone': 14, 'smartMode': 0, 'chargeMaxLimit': 2400, 'phaseSwitch': 1, 'rssi': -72, 'aiState': 0}, 'packData': [{'sn': 'FO4NHN4N2000314', 'packType': 5, 'socLevel': 74, 'state': 2, 'power': 9, 'maxTemp': 3011, 'totalVol': 4990, 'batcur': 65534, 'maxVol': 333, 'minVol': 333, 'softVersion': 4101, 'heatState': 0}, {'sn': 'FO4NHN4N2000468', 'packType': 5, 'socLevel': 74, 'state': 2, 'power': 10, 'maxTemp': 2991, 'totalVol': 5000, 'batcur': 65534, 'maxVol': 333, 'minVol': 333, 'softVersion': 4101, 'heatState': 0}]}
2025-08-03 12:02:27.341 DEBUG (MainThread) [custom_components.zendure_ha.manager] Updating Zendure data
2025-08-03 12:02:27.342 DEBUG (MainThread) [custom_components.zendure_ha.manager] Update device: SolarFlow 2400 AC (4MfAdFQc)
2025-08-03 12:02:27.342 DEBUG (MainThread) [custom_components.zendure_ha.manager] Finished fetching Zendure Manager data in 0.001 seconds (success: True)
2025-08-03 12:02:28.387 WARNING (MainThread) [homeassistant.components.mqtt.sensor] Ignoring invalid option received on topic 'ebusd/hmu/State', got '128', allowed: ready, heating, error, cooling, heating_water
2025-08-03 12:02:29.353 INFO (MainThread) [custom_components.zendure_ha.manager] Update p1: 7 power: 0.0 operation: 2 delta:6.875 stddev: 3.3330729064933458 fast: False
2025-08-03 12:02:29.353 INFO (MainThread) [custom_components.zendure_ha.manager] Start discharging with p1: 7
2025-08-03 12:02:29.353 INFO (MainThread) [custom_components.zendure_ha.manager] Update setpoint: 7 stateManagerState.DISCHARGING max power: 0
[... cut ...]
2025-08-03 12:02:43.404 WARNING (MainThread) [homeassistant.components.mqtt.sensor] Ignoring invalid option received on topic 'ebusd/hmu/State', got '128', allowed: ready, heating, error, cooling, heating_water
2025-08-03 12:02:43.684 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 270, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)

CommanderROR9, 2025-08-03:
Danke

harrymayr, 2025-08-03:
> Was den P1 Wert angeht...zu dem Zeitpunkt sollten so 1000W Überschuss vorhanden gewesen sein.

Das kann ich da nicht sehen. Und für den Test muss ja kein Überschuss vorhanden sein, deine Batterien sind ja zu ca. 74% voll, da kann ja auch eingespeist werden.
Schau dir deinen Sensor mal im Verlauf an, ob der wirklich den Überschuss bzw. Verbauch (negativ bzw. positiv) anzeigt.

CommanderROR9, 2025-08-03:
Ja, der Sensor zeigt Überschuss (negativ) und Verbrauch (Positiv) an. Ich habe ihn jetzt sogar in sensor.power_actual umbenannt.
Ich teste nochmal und logge mit.

CommanderROR9, 2025-08-03:
[custom_components.zendure_ha.manager] Update p1: -2741 power: 0.0 operation: 2 delta:-2741 stddev: 50 fast: True
2025-08-03 13:17:25.351 INFO (MainThread) [custom_components.zendure_ha.manager] Start charging with p1: -2741
2025-08-03 13:17:25.351 INFO (MainThread) [custom_components.zendure_ha.manager] Update setpoint: -2741 stateManagerState.CHARGING max power: 0

Ich denke da sieht man das Problem.

harrymayr, 2025-08-03:
OK, dann bitte mal folgendes an gleicher Stelle ändern, HA mit Debug Log neu starten und dann nach "power_capacity values" suchen

    def power_capacity(self, state: ManagerState) -> float:
        """Get the device capacity for state."""
        _LOGGER.info(f"power_capacity values - online: {self.online} electricLevel: {self.electricLevel.state} socSet: {self.socSet.state} minSoc: {self.minSoc.state} kWh: {self.kWh} powerAct: {self.powerAct}")
        if not self.online or self.electricLevel.state is None or self.socSet.state is None or self.minSoc.state is None:
            return 0.0

Bei mir steht dann sowas drin:

`INFO (MainThread) [custom_components.zendure_ha.device] power_capacity values - online: True electricLevel: 69 socSet: 100.0 minSoc: 11.0 kWh: 5.76 powerAct: -117.0`

harrymayr, 2025-08-03:
> OK, es war wohl nur ein kurzer Aussetzer. Leider geht aber nach wie vor nichts, trotz Anpassung der Datei. Die "power_capacity" taucht im Log nirgends auf.

Das ist sonderbar, das muss immer in dem Verbund auftreten:

```
2025-08-03 18:35:22.093 INFO (MainThread) [custom_components.zendure_ha.manager] Update p1: -2 power: -348.0 operation: 2 delta:176.0 stddev: 50 fast: True
2025-08-03 18:35:22.094 INFO (MainThread) [custom_components.zendure_ha.device] power_capacity values - online: True electricLevel: 98 socSet: 100.0 minSoc: 11.0 kWh: 5.76 powerAct: -348.0
2025-08-03 18:35:22.094 INFO (MainThread) [custom_components.zendure_ha.manager] Update setpoint: -350.0 stateManagerState.CHARGING max power: -1200
2025-08-03 18:35:22.094 DEBUG (MainThread) [custom_components.zendure_ha.manager] Set power for device: Hyper 2000 (11.52) to -350.0W
2025-08-03 18:35:22.095 INFO (MainThread) [custom_components.zendure_ha.devices.hyper2000] Update power Hyper 2000 => -350.0 state: ManagerState.CHARGING delta: 346.0
```

Im Zweifel nochmal die ganze Integration löschen (auch in HACS), HA neu starten, in HACS die Integration neu laden, HA neu starten, die Integration hinzufügen (Token gibt es in der App). Wenn dann immer noch nichts geht, die Zeile mit dem Logger nochmal einfügen und HA nochmal neu starten. Im Smart-Mode muss der Code aufgerufen werden. Gibt es ggf. einen Fehler in dem Bereich, der dann auch geloggt wird?

ralle6677, 2025-08-04:
@harrymayr Hast du schon Gelegenheit gehabt, das Log anzuschauen? Hatte ich dir gestern an die oben genannte Adresse geschickt. Danke dir!

harrymayr, 2025-08-04:
@ralle6677 Ja, ich hatte mal reingeschaut und auf den ersten Blick nichts auffälliges sehen können. Muss mich damit noch intensiver beschäftigen. Da mein Urlaub aber vorbei ist, geht das nur noch Abends 😉

closed 2025-08-05 by harrymayr