Zendure/Zendure-HA #533: Wrong Current while discharging with pre14

SukramJ, 2025-08-03:
For AB3000X the Current value is to high by a factor of 1000 while discharging
Instead of displaying 6.5A it shows 6500A.
Strangely it is correct while charging.
This is already a problem with pre13( didn’t test other) and 1.0.47!

harrymayr, 2025-08-03:
Seems to be a problem of the battery itself. Could be a type-cast error (interpretation of a signed value as unsigned) or really a factor.
In case of a typecast problem, the value is always positive, but should be negative for discharging.
Can you check, if the value matches with power and voltage? (Power / Voltage = Current)
If so, you can add the following lines in entity.py in entityUpdate line 180

        # update entity state
        if entity is not None and entity.platform and entity.state != value:
            match key:
                case "batcur":
                    if value < -20:
                        value /= 1000

            return entity.update_value(value)

If it is a typecast error then I would try this

        # update entity state
        if entity is not None and entity.platform and entity.state != value:
            match key:
                case "batcur":
                    if value  > 200:
                        value = (((value * 10) ^ 0x8000) - 0x8000) / 10

            return entity.update_value(value)

SukramJ, 2025-08-03:
The initial value of batcur while discharging is 65513.

Power is 111W
Voltage is 48,3V
So the expected value for current is 2,298A.

So my initial assumption of a factor of 1000 is wrong, but I don't know how to interpret 65513.

SukramJ, 2025-08-03:
A little bit better:

Now the initial value is correct: -2,4A
After 10s this value updates to 6.551,2 A

SukramJ, 2025-08-03:
This works:

```
            # update entity state
            if entity is not None and entity.platform and entity.state != value:
                match key:
                    case "batcur":
                        if value > 200:
                            value = (value ^ 0x8000) - 0x8000
                return entity.update_value(value)
            return True

        # update entity state
        if entity is not None and entity.platform and entity.state != value:
            match key:
                case "batcur":
                    if value > 200:
                        value = (value ^ 0x8000) - 0x8000
            return entity.update_value(value)

```

Initial value and follow ups are now correct.

closed 2025-08-05 by FireSon