Zendure/Zendure-HA #542: version  1.1.0-pre14 didn't work on hyper2000

FrColin, 2025-08-03:
Reinstalling from scratch the version pre14 , with the local MQTT mode , token and HA MQTT Addon
sensor are reporting values but smart mode did'nt work output power is always zero
program is unknown, offline on zendure app

**I got some sensors removed**:

connection_status , no report of bluetooth or other status cloud local, 
no more battery current sensors (_batcur)

**switching to cloud mode** make all sensors unavailable, offline on zendure app
the iot ...getALL message is always send on local mqtt

**reverting to local mode** didn't restore all the sensors 

**here some log output**

[home-assistant_zendure_ha_2025-08-03T19-27-48_zendure.log](https://github.com/user-attachments/files/21567847/home-assistant_zendure_ha_2025-08-03T19-27-48_zendure.log)

FireSon, 2025-08-04:
@FrColin can you try pre16, and create a logfile when mqtt logging is turned on?

FrColin, 2025-08-06:
Hi, @FireSon 
I discover that the cluster group now fuse group was not set ie unused.
I have only one hyper so i didn't pay attention to that but setting to "own phase" make the system running