Zendure/Zendure-HA #544: feature request

Zanooon97, 2025-08-04:
Hallo, ich weis ja nich wie gut die neue endladesteategie für die batterien ist, wie wäre es wenn man selber entscheiden kann ob man die inverter abwechselnd belasten will oder gleichzeitig durchgehend.

[image]

Zanooon97, 2025-08-04:
früher wurden meine zwei SF gleichzeitig zum entladen benutzt, jetzt wechseln sie ca  alle 10 min hin und her.
ich möchte wieder eine gleichzeitige entladung

Linus86529, 2025-08-04:
Aufgrund der vielen und schnellen Änderungen in den Versionen pre-4 bis pre-15 sowie der berechtigten Wünsche dazu, wie der Zendure Manager die neuen, stärkeren Geräte managen soll, habe ich Bedenken, dass Bewährtes aus dem Code verschwindet oder möglicherweise ungewollt unter den Tisch fällt.

Nach wie vor sollte es möglich sein, dass Cluster aus unterschiedlich performanten Geräten die „kleinen” Grundlasten zwischen 80 W und 400 W optimal aus mehreren Geräten gleichzeitig speisen können. Das funktioniert hervorragend in der Version 1.1.0 pre-3. Das möchte ich nicht vermissen.

Zanooon97, 2025-08-04:
@Linus86529 ja das meine ich doch, jedr sollte selber entscheiden können wie die last aufgeteilt wird, zb. mit einem dropdown menu ...

FireSon, 2025-08-04:
@Zanooon97 Will think about it once the current distribution is finalized.

closed 2025-08-04 by FireSon