Zendure/Zendure-HA #545: Neuer Hyper wird von Integration nicht erkannt/hinzugefügt

Rene0772, 2025-08-04:
Hallo,
bei mir läuft seit 2 Monaten die Integration erfolgreich und sehr gut mit meinen 4 Hypern und 8 Batterien.
Jetzt habe ich am Freitag endlich meinen letzten, fünten Hyper bekommen. Diesen habe ich genau wie die anderen in meinem primären Account registriert und dann meinem zweiten Zendure-Account für HomeAssistant freigegeben. In der App ist dieser auch einwandfrei zu sehen doch leider bekomme ich ihn in HomeAssistant nicht angezeigt.

Ich habe die Vermutung, dass er eine neuere Hardware-ID hat und nicht als Hyper 2000 erkannt wird.

Hier müsstet ihr euch bitte mal den Code für die Erkennung anschauen.

Danke und Grüße
René

Rene0772, 2025-08-04:
Hier noch ein Nachtrag. In den HA Logs habe ich jetzt folgendes gefunden:
2025-08-03 16:28:34.349 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Device Hyper2000_3.0 is not supported!

pulsarz, 2025-08-04:
This new revision is already supported in the prereleases.

Rene0772, 2025-08-04:
Is the new Hyper2000 V3 also supportet in the pre3 release?

Rene0772, 2025-08-04:
Okay, thanks for the information. Because of the Smart Managing is not working in new prereleases (see Issue #547) I will wait for a new final release