Zendure/Zendure-HA #546: power distribution logic (1.1.0 pre15)

CWILDMOUNTAIN, 2025-08-04:
Can someone explain the "rules" of the updated power distribution (use as few devices as possible- really happy about this btw).

It seems to me in my setup (3x Hyper) that not the one with the highest SOC is discharging but (maybe random) the first in the list. 

Is current still distributed dynamicly, based on the 'actual soc' for charging and discharging (as stated in the readme)?

Linus86529, 2025-08-04:
@FireSon Thx you for the explanation. What about the old devices, i.e. Hub1200 and Hub2000? Are these no longer supported by the new PRE(nn) versions? Or why do you only mention the current devices?

I use a cluster of a Hyper2000 and a Hub1200 which works very well with the 1.1.0-pre3. Both supply power to the house proportionally and that is perfect as it is implemented in the pre3. Can I assume that the versions after pre3 will still do the same? Or do the new versions only feed energy from the most fully charged batteries?

FireSon, 2025-08-04:
> [@FireSon](https://github.com/FireSon) Hi, where can I select the prefered power distribution?

There is no "prefered power distribution", only a power distribution 😉 This does not mean that there never will be one, but for now I am working on optimizing it. Personally I do not think there is a need for more than one, but I am happy to be proved wrong...

Zanooon97, 2025-08-04:
look at the left 5% less the soc and still dischargin?

https://github.com/user-attachments/assets/63e462cf-f021-4075-9e5b-63dd20636802

closed 2025-08-04 by FireSon