Zendure/Zendure-HA #548: Pre15 charging issue

pulsarz, 2025-08-04:
So switched to pre15 and found that when I have lets say 1600W of surplus solar 1 hyper chatges at 1200w and the other does nothing.

Switched back to pre14 and hyper 1 charges at 1200w and hyper 2 at 400w.

Discharging does seem fine on pre15.

FireSon, 2025-08-04:
@pulsarz Would it be possible that the hyper who did not work in pre15 was at its minSoc? In pre15 this was regarded as a device which was 'ready', which is true for discharging but not charging! Fixed that in the mean time...

FireSon, 2025-08-04:
> Switched back to pre15, turned on mqtt log and HA log. If I catch it again I will stop it and send you the logs.

I have checked the code again, and I think I have found the cause of the problem! One of my devices also did not load as expected...
It is indeed a problem specific to pre15! Will fix it in the next release. Thanks for the feedback!!

closed 2025-08-04 by FireSon