Zendure/Zendure-HA #558: Hyper 2000 - charging and discharging are not displayed

git849376, 2025-08-05:
Hi everyone, I just installed the Zendure HA integration (Version 1.1.0-pre16) in my home automation system. So far, all sensors are displayed. Unfortunately, the aggr_charge_total and aggr_discharge_total sensors are not. Therefore, no data is displayed in the energy dashboard. What could be causing this? Thanks in advance.

[image]

[image]

Geert2991, 2025-08-05:
Do you use the local mqtt option? Did you check the log?

harrymayr, 2025-08-05:
If you had installed an older version, then it is because the names had changed. Is the above screenshot from another panel you created in the past?

git849376, 2025-08-05:
> If you had installed an older version, then it is because the names had changed. Is the above screenshot from another panel you created in the past?

I had never installed the integration before. Today was the first installation with Version 1.1.0-pre16

harrymayr, 2025-08-05:
@6thGuest, @git849376 did make a restart of HA after adding the Integration?
If not, please try that. If this still not work, please enable debug logging, and restart HA again. After ~5min. stop debug logging and send me the downloaded file to harryzagithub@gmail.com. I will search for errors.

git849376, 2025-08-06:
the log file is full of the following entries

2025-08-06 09:20:27.019 ERROR (paho-mqtt-client-c4J7y5hs) [custom_components.zendure_ha.sensor] Unable to connect to Zendure float() argument must be a string or a real number, not 'NoneType'!
2025-08-06 09:20:28.056 ERROR (paho-mqtt-client-c4J7y5hs) [custom_components.zendure_ha.sensor] Unable to connect to Zendure float() argument must be a string or a real number, not 'NoneType'!
2025-08-06 09:20:28.058 ERROR (paho-mqtt-client-c4J7y5hs) [custom_components.zendure_ha.sensor] Unable to connect to Zendure float() argument must be a string or a real number, not 'NoneType'!
2025-08-06 09:20:29.085 ERROR (paho-mqtt-client-c4J7y5hs) [custom_components.zendure_ha.sensor] Unable to connect to Zendure float() argument must be a string or a real number, not 'NoneType'!
2025-08-06 09:20:29.087 ERROR (paho-mqtt-client-c4J7y5hs) [custom_components.zendure_ha.sensor] Unable to connect to Zendure float() argument must be a string or a real number, not 'NoneType'!
2025-08-06 09:20:30.116 ERROR (paho-mqtt-client-c4J7y5hs) [custom_components.zendure_ha.sensor] Unable to connect to Zendure float() argument must be a string or a real number, not 'NoneType'!
2025-08-06 09:20:30.118 ERROR (paho-mqtt-client-c4J7y5hs) [custom_components.zendure_ha.sensor] Unable to connect to Zendure float() argument must be a string or a real number, not 'NoneType'!
2025-08-06 09:20:31.196 ERROR (paho-mqtt-client-c4J7y5hs) [custom_components.zendure_ha.sensor] Unable to connect to Zendure float() argument must be a string or a real number, not 'NoneType'!
2025-08-06 09:20:31.198 ERROR (paho-mqtt-client-c4J7y5hs) [custom_components.zendure_ha.sensor] Unable to connect to Zendure float() argument must be a string or a real number, not 'NoneType'!
2025-08-06 09:20:32.265 ERROR (paho-mqtt-client-c4J7y5hs) [custom_components.zendure_ha.sensor] Unable to connect to Zendure float() argument must be a string or a real number, not 'NoneType'!
2025-08-06 09:20:32.267 ERROR (paho-mqtt-client-c4J7y5hs) [custom_components.zendure_ha.sensor] Unable to connect to Zendure float() argument must be a string or a real number, not 'NoneType'!
2025-08-06 09:20:34.345 ERROR (paho-mqtt-client-c4J7y5hs) [custom_components.zendure_ha.sensor] Unable to connect to Zendure float() argument must be a string or a real number, not 'NoneType'!
2025-08-06 09:20:34.347 ERROR (paho-mqtt-client-c4J7y5hs) [custom_components.zendure_ha.sensor] Unable to connect to Zendure float() argument must be a string or a real number, not 'NoneType'!
2025-08-06 09:20:35.398 ERROR (paho-mqtt-client-c4J7y5hs) [custom_components.zendure_ha.sensor] Unable to connect to Zendure float() argument must be a string or a real number, not 'NoneType'!
2025-08-06 09:20:35.400 ERROR (paho-mqtt-client-c4J7y5hs) [custom_components.zendure_ha.sensor] Unable to connect to Zendure float() argument must be a string or a real number, not 'NoneType'!
2025-08-06 09:20:36.412 ERROR (paho-mqtt-client-c4J7y5hs) [custom_components.zendure_ha.sensor] Unable to connect to Zendure float() argument must be a string or a real number, not 'NoneType'!
2025-08-06 09:20:36.414 ERROR (paho-mqtt-client-c4J7y5hs) [custom_components.zendure_ha.sensor] Unable to connect to Zendure float() argument must be a string or a real number, not 'NoneType'!
2025-08-06 09:20:37.465 ERROR (paho-mqtt-client-c4J7y5hs) [custom_components.zendure_ha.sensor] Unable to connect to Zendure float() argument must be a string or a real number, not 'NoneType'!
2025-08-06 09:20:37.469 ERROR (paho-mqtt-client-c4J7y5hs) [custom_components.zendure_ha.sensor] Unable to connect to Zendure float() argument must be a string or a real number, not 'NoneType'!
2025-08-06 09:20:38.489 ERROR (paho-mqtt-client-c4J7y5hs) [custom_components.zendure_ha.sensor] Unable to connect to Zendure float() argument must be a string or a real number, not 'NoneType'!
2025-08-06 09:20:38.491 ERROR (paho-mqtt-client-c4J7y5hs) [custom_components.zendure_ha.sensor] Unable to connect to Zendure float() argument must be a string or a real number, not 'NoneType'!
2025-08-06 09:20:39.523 ERROR (paho-mqtt-client-c4J7y5hs) [custom_components.zendure_ha.sensor] Unable to connect to Zendure float() argument must be a string or a real number, not 'NoneType'!
2025-08-06 09:20:39.525 ERROR (paho-mqtt-client-c4J7y5hs) [custom_components.zendure_ha.sensor] Unable to connect to Zendure float() argument must be a string or a real number, not 'NoneType'!
2025-08-06 09:20:40.549 ERROR (paho-mqtt-client-c4J7y5hs) [custom_components.zendure_ha.sensor] Unable to connect to Zendure float() argument must be a string or a real number, not 'NoneType'!
2025-08-06 09:20:40.551 ERROR (paho-mqtt-client-c4J7y5hs) [custom_components.zendure_ha.sensor] Unable to connect to Zendure float() argument must be a string or a real number, not 'NoneType'!
2025-08-06 09:20:41.583 ERROR (paho-mqtt-client-c4J7y5hs) [custom_components.zendure_ha.sensor] Unable to connect to Zendure float() argument must be a string or a real number, not 'NoneType'!
2025-08-06 09:20:41.585 ERROR (paho-mqtt-client-c4J7y5hs) [custom_components.zendure_ha.sensor] Unable to connect to Zendure float() argument must be a string or a real number, not 'NoneType'!
2025-08-06 09:20:42.614 ERROR (paho-mqtt-client-c4J7y5hs) [custom_components.zendure_ha.sensor] Unable to connect to Zendure float() argument must be a string or a real number, not 'NoneType'!
[... cut ...]
2025-08-06 09:23:50.912 ERROR (paho-mqtt-client-c4J7y5hs) [custom_components.zendure_ha.sensor] Unable to connect to Zendure float() argument must be a string or a real number, not 'NoneType'!
2025-08-06 09:23:51.922 ERROR (paho-mqtt-client-c4J7y5hs) [custom_components.zendure_ha.sensor] Unable to connect to Zendure float() argument must be a string or a real number, not 'NoneType'!
2025-08-06 09:23:51.928 ERROR (paho-mqtt-client-c4J7y5hs) [custom_components.zendure_ha.sensor] Unable to connect to Zendure float() argument must be a string or a real number, not 'NoneType'!

6thGuest, 2025-08-07:
Maybe stupid question - but is there an update function or do I have to install manually?

FireSon, 2025-08-07:
> Maybe stupid question - but is there an update function or do I have to install manually?

Not a stupid question at all! Once the 1.1.0 version is released it will show up in HA. As long as it is a pre-release, you have to manually install it...

6thGuest, 2025-08-07:
> Manual install means, reload another version in HACS

But the configuration has not to be done again? It is transferred from the former pre-release?

closed 2025-08-14 by FireSon

harrymayr, 2025-08-14:
> > Manual install means, reload another version in HACS
> 
> But the configuration has not to be done again? It is transferred from the former pre-release?

This depends a little bit from which version you update. If there is something in the configuration changed, than it is necessary to delete the old integration, restart HA and than add the new integration.