Zendure/Zendure-HA #561: New Solarflow firmware

Zanooon97, 2025-08-05:
Hi, need I to use this mqtt feature now or not?

[image]
[image]

z-master42, 2025-08-05:
Who came up with this shitty hint text? The device does not work with the HACS. HACS is just the store, like Google Play or the Apple Store. I think Zendure still doesn't understand how Home Assistant works.

6thGuest, 2025-08-05:
So is it possible JUST to monitor the 2400AC with this integration and not to "disconnect" it from Zenki or the Zendure-App?
I do not want to control the 2400AC via Home Assistant - I just want to display the data like SoC of the batteries, status (charging, discharging...)
The control of the 2400AC should still be done by Zendure KI (Zenki).
If yes - how can this be done?

z-master42, 2025-08-05:
> > So is it possible JUST to monitor the 2400AC with this integration and not to "disconnect" it from Zenki or the Zendure-App?
> > I do not want to control the 2400AC via Home Assistant - I just want to display the data like SoC of the batteries, status (charging, discharging...)
> > The control of the 2400AC should still be done by Zendure KI (Zenki).
> > If yes - how can this be done?
> 
> Yes. Simply install the integration and do nothing.

closed 2025-08-05 by FireSon

6thGuest, 2025-08-05:
> Install the current pre... version and use the cloud token from the current Zendure app to log in. Done.

Yeah - seems to work well :-)

Can you tell me why the sensors `sensor.solarflow_2400_ac_aggr_charge_total` and `sensor.solarflow_2400_ac_aggr_discharge_total`  only state `Unbekannt`

z-master42, 2025-08-05:
No, but it doesn't seem to be an isolated case at the moment #558.