Zendure/Zendure-HA #562: Hyper 2000 startet nicht mehr den Smart Modus Beta16

Wunni73, 2025-08-05:
Hallo ich habe Betta 16 und bei den geringen Preisen den Smart Modus ausgeschaltet .
Nachdem der Speicher voll ist von Netz Laden wollte ich wieder auf Smart Modus wechseln, 
aber es geht nicht . Cloud oder Lokal , es geht nicht .

Hi, I have a Betta 16 and, given the low price, I turned off Smart Mode.
After the storage was full from charging, I wanted to switch back to Smart Mode,
but it doesn't work. Cloud or local, it doesn't work.

Wunni73, 2025-08-05:
ich habe es gefunden. Geräte Cluster von unused in eigenes Kabel und eigene Phase eingestellt 

und schon geht es los 

@FireSon  I found it. I set the device cluster from unused to its own cable and phase.

And we go.

FireSon, 2025-08-05:
Great!! Can you still send me the error? Could help others as well...

FireSon, 2025-08-05:
@Kieft-C Instead of pointing out that it does not work, it might be better to create a logfile, preferably with the mqtt logging. I will take a look at it 😄

Wunni73, 2025-08-05:
KeyError: -1.0
2025-08-05 19:47:00.402 INFO (MainThread) [custom_components.zendure_ha.manager] Update p1: 657 power: 0 operation: 2 delta:657 stddev: 50 fast: True
2025-08-05 19:47:00.403 INFO (MainThread) [custom_components.zendure_ha.manager] Start discharging with p1: 657
2025-08-05 19:47:03.614 INFO (paho-mqtt-client-mqtt-home9992) [custom_components.zendure_ha.api] Topic: /         /log => {'messageId': 810862, 'logType': 2, 'timestamp': 1754416022, 'log': {'sn': '_x'_, 'params': [100, 3, 1000, 1000, 1000, 0, 5160, 5090, 5140, 0, 0, 0, 0, 0, 337, 335, 342, 0, 345, 340, 343, 0, 0, 0, 0, 0, 32772, 32772, 32772, 0, 8192, 0, 0, 0, 31, 29, 19, 0, 5052, 4761, 9062, 0, 4113, 4113, 8223, 0, 2, 0, 2, 1, 2217, 0, 5, 32, 0, 3303, 0, 52227, 3597, 0, 0, 14, 0, 0, 0, 0, 0, 2, 4, 250, 1200, 1, 1, 0, 0, 2, 0, 0, 1487, 28, 0, 0, 3, 0, 0, 0, 0, 0, 1, 3, 250, 50, 96, 17, 463, 255, 0, 0, 100, 10, 1, 3556, 5000, 0, 322, 63, 219, 0, 0, 0, 0, 0, 0, 1240, 800, 0, 0, 0, 0, 0]}}
2025-08-05 19:47:10.420 INFO (MainThread) [custom_components.zendure_ha.manager] Update p1: 652 power: 0 operation: 2 delta:652 stddev: 50 fast: True
2025-08-05 19:47:10.421 INFO (MainThread) [custom_components.zendure_ha.manager] Wait 10 sec for state change p1: 652
2025-08-05 19:47:17.159 INFO (paho-mqtt-client-mqtt-home9992) [custom_components.zendure_ha.api] Topic: /            /properties/report => {'messageId': 810887, 'product': 'solarFlowPro', 'timestamp': 1754416036, 'properties': {'electricLevel': 100}}
2025-08-05 19:47:17.161 INFO (paho-mqtt-client-mqtt-home9992) [custom_components.zendure_ha.api] Forwarding message from device   to cloud: / /properties/report => {'messageId': 810887, 'product': 'solarFlowPro', 'timestamp': 1754416036, 'properties': {'electricLevel': 100}, 'deviceId': ' ', 'isHA': True}
2025-08-05 19:47:20.355 INFO (MainThread) [custom_components.zendure_ha.manager] Update p1: 657 power: 0 operation: 2 delta:657 stddev: 50 fast: True
2025-08-05 19:47:20.975 INFO (MainThread) [custom_components.zendure_ha.manager] Update operation: 0 from: 2
2025-08-05 19:47:20.976 INFO (MainThread) [custom_components.zendure_ha.devices.hyper2000] Update power   => 0 state: ManagerState.IDLE delta: 0
2025-08-05 19:47:20.985 INFO (paho-mqtt-client-mqtt-home9992) [custom_components.zendure_ha.api] Topic: iot/ /function/invoke => {'arguments': [{'autoModelProgram': 0, 'autoModelValue': {'chargingType': 0, 'chargingPower': 0, 'freq': 0, 'outPower': 0}, 'msgType': 1, 'autoModel': 0}], 'function': 'deviceAutomation', 'messageId': 1, 'deviceKey': ' ', 'timestamp': 1754416040}
2025-08-05 19:47:21.408 INFO (paho-mqtt-client-mqtt-home9992) [custom_components.zendure_ha.api] Topic: /       /function/invoke/reply => {'messageId': 1, 'timestamp': 1754416040, 'function': 'deviceAutomation', 'output': 'success', 'success': 1}
2025-08-05 19:47:28.245 INFO (MainThread) [custom_components.zendure_ha.manager] Update operation: 2 from: 0
2025-08-05 19:47:30.389 INFO (MainThread) [custom_components.zendure_ha.manager] Update p1: 653 power: 0 operation: 2 delta:653 stddev: 50 fast: True
2025-08-05 19:47:30.390 INFO (MainThread) [custom_components.zendure_ha.manager] Start discharging with p1: 653
2025-08-05 19:47:40.425 INFO (MainThread) [custom_components.zendure_ha.manager] Update p1: 657 power: 0 operation: 2 delta:657 stddev: 50 fast: True
2025-08-05 19:47:40.426 INFO (MainThread) [custom_components.zendure_ha.manager] Wait 10 sec for state change p1: 657
2025-08-05 19:47:50.351 INFO (MainThread) [custom_components.zendure_ha.manager] Update p1: 652 power: 0 operation: 2 delta:652 stddev: 50 fast: True
KeyError: -1.0
2025-08-05 19:47:57.325 DEBUG (MainThread) [custom_components.zendure_ha.manager] Updating Zendure data
2025-08-05 19:47:57.325 INFO (MainThread) [custom_components.zendure_ha.manager] Found Zendure Bluetooth device: <BluetoothServiceInfoBleak name=ZenE1_74 address=94:C9:60:D8:AE:56 rssi=-71 manufacturer_data={   .'} service_data={} service_uuids=['nn'] source=A0:85:E3:4E:CA:1A connectable=True time=1637965.808566885 tx_power=None raw=b'\x02\x01\x06\x03\x19\x00\x00\x05\x03\x02\xa03"\x11\xffEE1LHMH1M371374.'>
2025-08-05 19:47:57.325 DEBUG (MainThread) [custom_components.zendure_ha.manager] Update device: 
2025-08-05 19:47:57.329 DEBUG (MainThread) [custom_components.zendure_ha.manager] Finished fetching Zendure Manager data in 0.004 seconds (success: True)
2025-08-05 19:47:57.333 INFO (paho-mqtt-client-mqtt-home9992) [custom_components.zendure_ha.api] Topic: iot/  /properties/read => {'properties': ['getAll']}
2025-08-05 19:47:57.489 INFO (paho-mqtt-client-mqtt-home9992) [custom_components.zendure_ha.api] Topic: /  /properties/read/reply => {'messageId': 134218099, 'timestamp': 1754416076, 'success': 1, 'properties': {'getAll': 1}}
2025-08-05 19:47:57.808 INFO (paho-mqtt-client-mqtt-home9992) [custom_components.zendure_ha.api] Topic: / /log => {'messageId': 810965, 'logType': 2, 'timestamp': 1754416076, 'log': {'sn': '   ', 'params': [100, 3, 1000, 1000, 1000, 0, 5160, 5090, 5140, 0, 0, 0, 0, 0, 337, 334, 342, 0, 345, 340, 343, 0, 0, 0, 0, 0, 32772, 32772, 32772, 0, 8192, 0, 0, 0, 31, 29, 19, 0, 5052, 4761, 9062, 0, 4113, 4113, 8223, 0, 2, 0, 2, 1, 2216, 0, 5, 32, 0, 3301, 0, 52227, 3592, 0, 0, 14, 0, 0, 0, 0, 0, 2, 4, 250, 1200, 1, 1, 0, 0, 2, 0, 0, 1487, 28, 0, 0, 3, 0, 0, 0, 0, 0, 1, 3, 250, 50, 96, 17, 463, 255, 0, 0, 100, 10, 1, 3551, 4999, 0, 322, 63, 219, 0, 0, 0, 0, 0, 0, 1240, 800, 0, 0, 0, 0, 0]}}
[... cut ...] erties': {'masterSwitch': 1, 'electricLevel': 100, 'wifiState': 1, 'faultLevel': 0, 'socSet': 1000, 'solarInputPower': 0, 'packInputPower': 0, 'outputPackPower': 0, 'outputHomePower': 0, 'gridInputPower': 0}, 'deviceId': '', 'isHA': True}
2025-08-05 19:48:00.417 INFO (MainThread) [custom_components.zendure_ha.manager] Update p1: 651 power: 0 operation: 2 delta:651 stddev: 50 fast: True
2025-08-05 19:48:00.418 INFO (MainThread) [custom_components.zendure_ha.manager] Start discharging with p1: 651

CommanderROR9, 2025-08-05:
This appears to be a somewhat tricky issue. Mine was working for a bit this afternoon,  but then went haywire again and I had to stop it. I can't figure it out. I sent some Logs earlier today, hopefully you can see whatever is wrong. For some Reason my SF2400 wants to send more Power to the House than it can accept, so essentially it foreces some feed-in to the grid. And it also sometimes gets "stuck" wither Charging or Discharging. 
Local MQTT is now possible though, thanks to new Firmware, so maybe that will help somehow.

Wunni73, 2025-08-06:
@FireSon @CommanderROR9 Hallo ich habe version 17 updatet es ist auf 500 W Einspeisung hängen geblieben.
Connection nicht verbunden ?? es hat aber alle werte Aktualisiert ??  
Smart Mode 
Smart Mode -> Off er hat nach 1 Minute umgeschalten Auf Netz Ladung = Alles OK 
Smart Mode ON keine reaktion  (Connection  Status nicht verbunden ?? )

Sicherungs Gruppe war wieder auf nicht verwendet !!

umgestellt auf eigener Stromkreis eigene Phase   - und schon geht Alle  Connection verbunden Smart Mode startet

Hello, I updated to version 17, but it got stuck on 500 W input.
Connection not connected?? But it updated all values??
Smart Mode on
Smart Mode -> Off, it switched to mains charging after 1 minute = everything OK.
Smart Mode ON, no response (Connection status not connected??)

cluster group was set to unused again!!

Switched to separate circuit, separate phase - and now all connections are connected, Smart Mode starts. it work Top

CommanderROR9, 2025-08-06:
Good to hear that it works, but keep an eye on it. I had some success with Version pre-16 yesterday, but it broke down again after a while. My battery is almost full today and I will get an additional 2.88kWh battery pack later today, so my testing conditions aren't ideal.
For me the real test will come, when the system switches to "feed in" mode, because that's where I had problems with pre-16. It worked fine, but didn't zero out the input/output but instead kept feeding electricity into the grid somewhere between 150-200W which of course doesn't make sense.

Wunni73, 2025-08-06:
you have to choose the Fusegroup otherwise it won't work

CommanderROR9, 2025-08-06:
> you have to choose the Fusegroup otherwise it won't work

I know..I have been running with the fuse group settings enabled, even before they were called fuse group😉

Kieft-C, 2025-08-08:
@FireSon 

Did all what u say.

8. "Make sure your hyper has the latest firmware (master: v2.1.30)" --> Check it is
9.  "Check if your hyper has a Bluetooth Mac address, if not make sure this is available. The Integration scans each 90 seconds for the mac address."  --> Check, see the Mac Bluetooth adress
10. "Ceck your connection status, it should say connected. " --> is connected
13.  "Check if your p1 meter gives the correct reading (from the logfile this seemed ok)" -> feels right

14.  "Select the ZendureManager operation "Manual Power" and enter for instance 300w in the Zendure Manager Manual Power. After say 10 second, set manual power to 400w" --> no reaction

16. "If the hyper is not reacting, send me the logfile of the complete process until this point" --> no reaction

I only get power out of the Hyper when I try "AC - Ausgangsmodus" with the slider direkt in the hyper menu.
But with the zendure manager, no reaction.

File here:

[home-assistant_zendure_ha_2025-08-08T04-30-26.109Z.log](https://github.com/user-attachments/files/21676986/home-assistant_zendure_ha_2025-08-08T04-30-26.109Z.log)

Kieft-C, 2025-08-08:
I have to wait, that the bypass will off and not work, so I guess in the late evening/early night I will do this.
Fingers crossed. Thx a lot @FireSon

Kieft-C, 2025-08-08:
@FireSon 
testet Version 1.1.1 with no luck
Here the log file.

[home-assistant_zendure_ha_2025-08-08T20-32-24.659Z.log](https://github.com/user-attachments/files/21691318/home-assistant_zendure_ha_2025-08-08T20-32-24.659Z.log)

FireSon, 2025-08-09:
@Kieft-C Is possible that your HA system is mot very powerful? It seems that not all entities are initialized when they are used. I have pushed some changes, which should take care of that, can you test it?

Kieft-C, 2025-08-09:
> @Kieft-C Is possible that your HA system is mot very powerful? It seems that not all entities are initialized when they are used. I have pushed some changes, which should take care of that, can you test it?

It is a HA Green System. Should have enough power and I have nothing else Running.
Will Check RAM and CPU later.

Kieft-C, 2025-08-09:
Processor usage is 4-6% in idle mode while pre3 is running (and other small stuff)
from 4gb is 1,3gb in usage.

So it should be powerful enough. i hope and guess?

FireSon, 2025-08-09:
@Kieft-C Yes I allready did some changes, so please test the latest version of the master branch, and send me the log file 😄

Kieft-C, 2025-08-09:
> [@Kieft-C](https://github.com/Kieft-C) Yes I allready did some changes, so please test the latest version of the master branch, and send me the log file 😄

I try it. With no luck. sadly
Maybe u can also name the title always when u change something, that i'm sure that i got a changed code in the system.
I downloaded it and it shows me v1.1.1 again.

i get to debug logs, maybe that helps?
I also try different things in fuse group and dischaching etc.
But nothing work, manual mode and smart mode not react. 

[home-assistant_zendure_ha_2025-08-09T17-08-38.392Z.log](https://github.com/user-attachments/files/21699950/home-assistant_zendure_ha_2025-08-09T17-08-38.392Z.log)

[home-assistant_zendure_ha_2025-08-08T20-32-24.659Z.log](https://github.com/user-attachments/files/21699951/home-assistant_zendure_ha_2025-08-08T20-32-24.659Z.log)

FireSon, 2025-08-09:
@Kieft-C Ai there was a problem with my update, unfortunately you tested the wrong code. Could you do it again? I updated the version to 1.1.1-pre1

Kieft-C, 2025-08-09:
> [@Kieft-C](https://github.com/Kieft-C) Ai there was a problem with my update, unfortunately you tested the wrong code. Could you do it again? I updated the version to 1.1.1-pre1

Will try it later at the day. thx for your time.

Kieft-C, 2025-08-09:
> [@Kieft-C](https://github.com/Kieft-C) Ai there was a problem with my update, unfortunately you tested the wrong code. Could you do it again? I updated the version to 1.1.1-pre1

I got 1.1.1-pre1 uploaded, test it. with no luck. Here the Log.

[home-assistant_zendure_ha_2025-08-09T20-02-58.384Z.log](https://github.com/user-attachments/files/21701325/home-assistant_zendure_ha_2025-08-09T20-02-58.384Z.log)

harrymayr, 2025-08-09:
Maybe you are a little to impatient. I see 2 errors wrt to the integration, because of a missing or not yet initialized entity socStatus. But later on, this entity was shown up. But then the log end at 22:02:42 right after the BLE command for the local MQTT was send (and you changed to Smart Matching).

FireSon, 2025-08-09:
@Kieft-C I think Harry is right, I initialized the state of the socLimit now during creation, just to see if that fixes the problem. Compared to the previous log, several errors have disappeared, so that is good 😄 pre2 should be the version

Kieft-C, 2025-08-09:
ungeduldig, relativ. ;-)
-pre3 version startet der Smartmatch mode halt binnen 10sec und läuft sofort sauber.  

v1.1.1-pre2 funktioniert bei mir der manual mode. Aber leider weiterhin nicht der smartmatch mode.

Hier das log. Hab HA neu gestartet und Zeit gegeben mal bisschen zu schauen.

@harrymayr die verwechselung mit dem P1 sensor, passiert dadurch, das ich anfangs eine integration aus HACS neu geladen hatte.
Hab jetzt erst gelernt wie man via Samba direkt dateien+ordner hochladen kann. hat meinen workflow extrem beschleunigt. ;-)

[home-assistant_zendure_ha_2025-08-09T21-29-05.005Z.log](https://github.com/user-attachments/files/21702358/home-assistant_zendure_ha_2025-08-09T21-29-05.005Z.log)

harrymayr, 2025-08-09:
Schon wieder verwechselt?

[image]

FireSon, 2025-08-09:
HI @Kieft-C I think Harry is right! Also if the manual power can update the device, once the p1 meter value is availble the smartmode should work as well 😄

Kieft-C, 2025-08-09:
> Schon wieder verwechselt?
> <img alt="Image" width="481" height="92" src="https://private-user-images.githubusercontent.com/104222469/476314116-aa0f79ea-445b-443c-984c-b2e11fd1fa6b.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTQ3NzU5NjMsIm5iZiI6MTc1NDc3NTY2MywicGF0aCI6Ii8xMDQyMjI0NjkvNDc2MzE0MTE2LWFhMGY3OWVhLTQ0NWItNDQzYy05ODRjLWIyZTExZmQxZmE2Yi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwODA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDgwOVQyMTQxMDNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xMGUyZWI2YTM2ZGQ1Zjg3MmI3YzQxNzMzMDE0OTgzMTY2NDYyZDRkYzVhNDJmYmUzYThmNTUyZjBhZGU5NDk4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.59EErH9d5UwRu-DdR37UtHSFlZfQvaHzJ1mI1mZ-h_M">
> 
> Keine Updates vom P1 sensor, daher auch kein Smart Matching?

Ganz sicher (eigentlich) nicht. Das ist ja merkwürdig. Hab ein Txt dokument, wo ich bei jeder version & einrichtunges raus kopiere und eintrage. Step by step. Das er den p1 nicht nimmt und ignoriert könnte keine option sein? Ich bin irritiert.
Dies scheint nach dem HA Neustart passiert zu sein. Ich probier es jetzt nochmals. :)

Kieft-C, 2025-08-09:
[image]

Ja, es scheint irgend was mit dem Power sensor nach dem Neustart bzw. bei Initalisierung zu sein.
Screenshot siehe wie ich es eingerichtet hab.

[home-assistant_zendure_ha_2025-08-09T21-55-14.891Z.log](https://github.com/user-attachments/files/21702450/home-assistant_zendure_ha_2025-08-09T21-55-14.891Z.log)
Ein Log vor dem Neustart und eins nach dem Neustart:

[home-assistant_zendure_ha_2025-08-09T22-00-14.556Z.log](https://github.com/user-attachments/files/21702452/home-assistant_zendure_ha_2025-08-09T22-00-14.556Z.log)

Kieft-C, 2025-08-09:
> Lösch mal die Integration, starte HA neu und richte sie dann erneut ein. Ich sehe immer noch power_actual.

Will do. Aber erstes log file:
"2025-08-09 23:49:32.467 DEBUG (MainThread) [custom_components.zendure_ha.manager] Finished fetching Zendure Manager data in 0.000 seconds (success: True)
2025-08-09 23:50:30.580 DEBUG (MainThread) [custom_components.zendure_ha] Updating Zendure config entry: 01K28BZSW2T3H1V0PZJZBX8APT
2025-08-09 23:50:30.580 DEBUG (MainThread) [custom_components.zendure_ha.manager] Updating P1 meter to: sensor.shellypro3em_a0dd6ca0b780_total_active_power
2025-08-09 23:50:30.581 DEBUG (MainThread) [custom_components.zendure_ha] Unloading Zendure config entry: 01K28BZSW2T3H1V0PZJZBX8APT
2025-08-09 23:50:30.582 DEBUG (MainThread) [custom_components.zendure_ha.manager] Updating P1 meter to: None
"

Kieft-C, 2025-08-09:
Jup, es sieht stark danach aus, dass er die eintragung des p1 sensors bei der einrichtung nicht schluckt und ignoriert.
Hab System neu gestartet, integration 1:1 hochgeladen, neu geladen und eingerichtet. Und dann verschwindet im log immer der power sensor.

[home-assistant_zendure_ha_2025-08-09T22-19-35.156Z.log](https://github.com/user-attachments/files/21702529/home-assistant_zendure_ha_2025-08-09T22-19-35.156Z.log)

harrymayr, 2025-08-09:
Dann mach dir als Workaround einen Template Sensor mit Namen power_actual, der die Werte deines Shelly verwendet.

[image]

[image]

harrymayr, 2025-08-10:
Hab den Bug gefunden, ist leider in der 1.1.1 noch drin:
Die Zeile 135 in manager.py durch die folgende ersetzen:
`        self.update_p1meter(self.config_entry.data.get(CONF_P1METER, "sensor.power_actual"))
`

[image]

Kieft-C, 2025-08-10:
[Version 1.1.2-pre2](https://github.com/Zendure/Zendure-HA/releases/tag/1.1.2-pre2)
Scheint bei mir nun sauber mit dem Hyper zu laufen. Und BLE Mac adresse des hypers wurde auch direkt wieder angezeigt.
Für mich wurde dieses problem damit für den hyper erledigt.

closed 2025-08-10 by harrymayr