Zendure/Zendure-HA #567: Questions to the Settings and Sensors in ZendureHA

DerUnwissende, 2025-08-06:
Hallo zusammen,

erstmal ganz lieben und herzlichen Dank die viele, viele Arbeit, die in der Integration der Zendure-HA steckt an alle, die dazu beigetragen haben.
Ich möchte die Integration aber auch besser verstehen und daraus mein Handeln besser ableiten.  

Info vorweg:
Ich bezeichne mich als Anfänger, der angefixt ist...
Nur damit man mein Ziel besser versteht: Ich möchte sehr gerne der Nulleinspeisung nahe kommen....
Ich habe PV-Module mit Hoymiles Wechselrichern (2x) und diese lese ich aktuell mit OpenDTU aus. Das funktioniert soweit.
Des weiteren habe ich im Zählerschrank 
1x EcoTracker IR
1x SmartMeter Zendure 3CT 
eingebaut.
Allerdings verwende ich den Smartmeter Zendure 3CT nicht, da aktuell dieser in der Integration nicht angezeigt wird. Darum nehme ich als Verbrauchsüberwacher den EcoTracker IR.

 

Ich habe bei mir aktuell als Pre-116 laufen und habe allerdings noch Verständnisprobleme zu den Einstellungen und der Bedeutung einiger Sensoren.
Dazu habe ich mal versucht meine Fragen in einem Screenshot meiner Installation mit Icons zu versehen.

grüner Haken = ist klar
rotes Fragezeichen ? ist mir nicht klar, was damit gemeint ist
und direkte Eintragungen im Screenshot um meine Problematiken aufzuzeigen

Gibt es irgendwo vielleicht eine Doku, die die Variablen / Zustände / Angaben nachlesbar machen? Dann hätten sich viele Punkte wahrscheinlich schon gelegt.

Wenn wir oben anfangen...

- Mir ist nicht klar, was ich mit dem Umschalten zwischen AC-Eingangsmodus / AC-Ausgangsmodus "anrichte" bzw. bewirke.   Warum muss man das dort angeben?
- Mir ist nicht klar, warum bei mir aktuell SolarFlow 2400 Ausgang auf 44 W steht. Was macht der Wert, bzw diese sich selbst einstellende Regelung? Wo fiesst das ein?
-   ist es richtig, dass der Bypass Modus auf "immer aus" steht?
-  Muss die SolarFlow 2400AC connection auf Cloud stehen, oder auf ZenSDK? (wenn möglich möchte ich nicht über Cloud, sondern hausintern kommunizieren) Ist ZenSDK schon implementiert? Muss ich dazu dann irgendwo noch weiteres umstellen? Mir ist das leider völlig unklar. Sagen kann ich nur, dass momentan sich die Regler automatisch verschieben, somit ist wahrscheinlich die Kommuniktation über Cloud ok. Aber ich habe Sorge, wenn ich probeweise auf ZenSDK umstelle, dass ich mir dann was kaputt schiesse.
- Ich habe für SolarFlow 2400 AC SOC als unteren Wert 10% angegeben. Damit will ich eigentlich erreichen, dass der Akku nicht bis auf 0 runter geht. Soll doch für Akkus nicht gut sein... Allerdings scheint der Wert ignoriert zu werden, denn unter Ladezustand Battery sieht man aktuell 6%, aber 2 oder 3% sind auch keine Seltenheit. Oder verstehe ich den Zusammenhang falsch?
- Ich vermute, dass Battery AB3000X und SolarFlow 2400 AC Batterie das Gleiche ist?
- Ist es richtig vermutet, dass SF 2400 AC Ausgangsleistung der Wert ist, der vom Akku ins Haus eingespeist wird?
- Ist es richtig vermutet, dass SF 2400 AC Eingangsleistung der Wert ist, der den Überschuss im Haus in den Akku schiebt?
- Was muss ich mir weiter unten unter SF 2400 AC solarPower1-6 vorstellen? Gilt das evt. nicht für mein Gerät?

Es wäre toll, wenn Ihr mir hier vielleicht etwas helfen könntet Licht ins Dunkle zu bringen.  Danke dafür im Voraus.

PS: @FireSon ich vermute, dass das Smartmeter 3CT von Zendure ebenfalls noch kommt, damit man das für die Thematik Nulleinspeisung nutzen kann?

----------------
here again the same issue in english:
Hello everyone,

First of all, I would like to express my sincere thanks to everyone who contributed to the integration of Zendure-HA for all the hard work they put into it.
However, I would also like to better understand the integration so that I can better derive my actions from it.  

Some background information:
I consider myself a beginner who is hooked...
Just so you can better understand my goal: I would very much like to get close to zero feed-in....
I have PV modules with Hoymiles inverters (2x) and I am currently reading them out with OpenDTU. That works so far.
I have also installed the following in the meter cabinet 
1x EcoTracker IR
1x SmartMeter Zendure 3CT .

However, I am not using the Smartmeter Zendure 3CT, as it is currently not displayed in the integration. That is why I am using the EcoTracker IR as a consumption monitor.

I currently have Pre-116 running on my system, but I still have some issues understanding the settings and the meaning of some sensors.
I have tried to illustrate my questions with icons in a screenshot of my installation.

Green check mark = clear
Red question mark ? I don't understand what this means
And direct entries in the screenshot to illustrate my problems

Is there perhaps a document somewhere that explains the variables/states/information? Then many of my questions would probably already be answered.

Let's start at the top...
- I am not sure what switching between AC input mode and AC output mode does or what effect it has. Why does this need to be specified there?
- I am not sure why my SolarFlow 2400 output is currently set to 44 W. What does this value do, or what does this self-adjusting control do? Where does it come into play?
- Is it correct that the bypass mode is set to "always off"?
- Does the SolarFlow 2400AC connection have to be set to Cloud or ZenSDK? (If possible, I would prefer to communicate internally rather than via the cloud.) Has ZenSDK already been implemented? Do I need to change anything else? Unfortunately, I am completely unclear about this. All I can say is that the controllers are currently shifting automatically, so communication via Cloud is probably okay. But I'm worried that if I switch to ZenSDK on a trial basis, I'll break something.
- I have set the lower value for SolarFlow 2400 AC SOC to 10%. My intention is to prevent the battery from going down to 0. That's not good for batteries... However, the value seems to be ignored, because under Battery Charge Status you can currently see 6%, but 2 or 3% is also not uncommon. Or am I misunderstanding the connection?
- I assume that Battery AB3000X and SolarFlow 2400 AC Battery are the same thing?
- Is it correct to assume that SF 2400 AC output power is the value that is fed from the battery into the house?
- Is it correct to assume that SF 2400 AC input power is the value that pushes the surplus in the house into the battery?
- What does SF 2400 AC solarPower1-6 mean? Does this not apply to my device?

It would be great if you could help me shed some light on this.  Thank you in advance.

PS: @FireSon, I assume that the Smartmeter 3CT from Zendure is also coming, so that it can be used for zero feed-in?

[image]

DerUnwissende, 2025-08-06:
Hej @CommanderROR9 ...
whauu... das ist ja richtig toller Input für mich... Sei dafür herzlichst bedankt...! 👍 

Durch Deine Hinweise habe ich die Firmware geupdatet und bin jetzt auch auf Pre-117
Kannst du mir einen Link oder Tipp geben, wie ich an den API-Schlüssel komme? Da stehe ich gerade auf dem Schlauch 😕 
Und ich bin mir aktuell auch nicht sicher, was du mit Fuse-Group meinst... Oder ist das Inhalt der pre-117? Ich habe zwar geupdatet, aber noch nicht reingeschaut.

FireSon, 2025-08-06:
@DerUnwissende Es gibt auf der [GitHub](https://github.com/Zendure/Zendure-HA/) seite mehere Info's zu die verschiedene Fragen. 

[Installation](https://github.com/Zendure/Zendure-HA/wiki/Installation)
[Fuse Group](https://github.com/Zendure/Zendure-HA/wiki/Fuse-Group) 

Wenn da etwas unklar ist, konnen wir das gerne erweitern 😄

CommanderROR9, 2025-08-06:
> Hej [@CommanderROR9](https://github.com/CommanderROR9) ... whauu... das ist ja richtig toller Input für mich... Sei dafür herzlichst bedankt...! 👍
>
> Durch Deine Hinweise habe ich die Firmware [...]

Api Key gibt es in der App. Ich glaube unter Profil war es. Fuse Group ist in der 117 zu finden.

FireSon, 2025-08-22:
Think I can close this issue now, if anything is unclear let me know

closed 2025-08-22 by FireSon