Zendure/Zendure-HA #568: WARNUNG aus HA Protokollle

Kieft-C, 2025-08-06:
Logger: homeassistant.util.logging
Quelle: util/logging.py:74
Erstmals aufgetreten: 20:16:16 (1 Vorkommnis)
Zuletzt protokolliert: 20:16:16

Module custom_components.zendure_ha.zendurebase is logging too frequently. 200 messages since last count

FireSon, 2025-08-06:
@Kieft-C That is from a pretty old version. But it is no problem, thanks for the feedback!

closed 2025-08-06 by FireSon

Kieft-C, 2025-08-06:
But how that could be? I work with -pre3 fresh installed or -p14 / -pre15 / -pre16.
May thats also a problem from my HA? I did not get the integration in total clean deinstalled?

I delete the download and delete the integration in devices. Then start new from HACS with normal steps.
That should renew everything, or?