Zendure/Zendure-HA #578: Freezing behavior with 1.0.47 since one week (SF1200)

surfer1264, 2025-08-07:
Ich benutze die 1.0.47 (noch) (SF1200), seit Juni?)
Ich benutze den "Manuellen Leistungsmodus", den ich mit meiner eigenen Regel-Engine im HA steuere

Seit einer Woche (Ende Juli) bleibt die Regelung häufig stehen
.....weil 
der Durchgriff vom HA auf die die "number.zendure_manager_manual_power" offensichtlich hängen bleibt
Gleichzeitig lassen sich auch die Ladelimits nicht mehr anpassen
Gleichzeitig habe ich auch keinen Durchgriff auf das Output-Limit oder die Bypass Entität

Lesende Zugriffe funktionieren gleichwohl alle. Änderungen an der App werden im HA gespiegelt
nur Änderungen im HA werden "irgendwann" ignoriert

Ein Restart des HA fixed das Problem für eine gewisse Zeit....

Kann man diese "Hänger" irgendwie erkennen?
Kann man den Zendure Manager Soft restarten...ohne HA zu resetten?

DANKE für Eure Gedanken (ansonsten weiter so....DANKE für Euren Aufwand und Engagement)

**********************
I am (still) using 1.0.47 (SF1200), since June?
I use “Manual Power Mode,” which I control with my own rule engine in HA

For a week now, the control has been stuck
...because 
the HA's access to “number.zendure_manager_manual_power” is obviously stuck
At the same time, the load limits can no longer be adjusted
At the same time, I also have no access to the output limit or the bypass entity

Read accesses all work, however. Changes made to the app are mirrored in HA
only changes in HA are ignored “at some point”

Restarting HA fixes the problem for a certain amount of time....

So I am also stuck

DerUnwissende, 2025-08-07:
Moin, 
da hat sich wohl einiges geändert.... Zendure hat einiges umgestellt, Updates in der Appversion, Firmware und MQTT-Dialogen.
Vielleicht gibst Du der aktuellen 1.047 - pre119 eine Chance. Bei mir läuft das mit lokalem MQTT ziemlich gut aktuell...

z-master42, 2025-08-08:
Mit der App hat das nichts zu tun.
Firmware-Updates gab es für die Hubs schon lange nicht mehr und wird es sehr wahrscheinlich auch nicht mehr geben.
An den MQTT-Topics hat sich auch nichts geändert.

Die Version 1.1.0 ist seit kurzen raus. Probier's mal damit.

closed 2025-08-08 by surfer1264