Zendure/Zendure-HA #585: 1.1.0 not working with SF2400

gowtah, 2025-08-08:
Hello and thanks for all the good work so far, I've been a happy user of previous versions without issues for a couple months.

Only now with 1.1.0 I'm having a couple breaking issues.

Using SF2400 with latest firmware 1.0.6, the local MQTT part of the integration does not seem to work. After setting up the integration with the API key and MQTT infos (using the HA mqtt broker addon), I do not get any new connection to local MQTT and the local option in the dropdown box is not there (only cloud and ZenSDK are available). The solarflow connects to the MQTT broker if I setup the same credentials in the zendure app, but that does not change anything to the dropdown menu.

Also, now even when set to "cloud" connection, any change of the Zendure manager "mode" stops the solarflow. Smart modes do not work at all. Manual power does not work either, no matter the wattage I type in. Basically any action with the zendure manager drop down menu after setting up manual charging in the zendure app puts the solarflow to sleep.

I can see nothing of interest in the logs, and I have not found yet how to enable the debugging mode.

gowtah, 2025-08-08:
As a follow up, here's an example of the behaviour.

SF set to AC in at 750W in the zendure app. Zendure manager on stop, the SF goes to sleep. Nothing I do in the zendure manager affects the SF afterwards.

SF set to AC in at 750W in the zendure app again. Switch zendure manager to smart coupling, nothing happens. Switch to smart charge, the SF goes to sleep after charging at 30W for a few seconds. Nothing I do in the Zendure manager affects the SF afterwards.

I think this behaviour started after I tried using the zendure app local MQTT setting to check if it worked that way since nothing showed up in HA's MQTT using the integration. Could it be that setting up MQTT in the zendure app breaks the cloud somehow? I have removed the infos and restarted several time since.

gowtah, 2025-08-08:
Ok so I tried this and it did not work at first. But then it seems it started working after I changed the fuse group to its own fuse group rather than leaving the default "unused no fuse group" (using the French locale sorry I don't have the specific english strings).
My battery is 100% charged right now so the test isn't the same as earlier today but the smart coupling did adjust the discharge when I put on an electric kettle.

Still nothing showing up on the local MQTT though.

gowtah, 2025-08-09:
Thanks for the support and the clarification, everything seems to be working fine using the zenSDK option. Perhaps you should write that clearly in the documentation because I bet I am not the only one who was confused about it :)

Keep up the good work!

CommanderROR9, 2025-08-09:
Keep monitoring it though. Mine worked fine for about 20 hours, then when the time came to switch to "discharge mode" again my SF2400AC stayed in idle mode, fully charged, until I manually jiggled a few entities and woke it up again.
Not sure exactly what the issue was though.