Zendure/Zendure-HA #586: Remaining discharge time is displayed while charging ??

Wunni73, 2025-08-08:
Verbleibende Entladezeit wird beim Laden angezeigt ??

Remaining discharge time is displayed while charging ??

z-master42, 2025-08-08:
OK. Entladen Zeit funktioniert bei mir. Ladezeit funktioniert schon seit Monaten nicht mehr. Da gab es diese Integration noch gar nicht. Hat Zendure irgendwann mal abgestellt, steht seit dem aus 999 h. Wenn zwischen Laden und Entladen gewechselt wird, bleibt die Dauer aber auch auf dem letzten Wert stehen. Ist zumindest bei mir so (Hub 2000).

Wunni73, 2025-08-08:
@z-master42 habe einen Hyper und der Wert kam immer bis gestern. 
Wen ich Lade dan war 999 dort gestanden.
Für die Ladezeit gab es ab pre 5 eine Formel die es errechnet hat.

z-master42, 2025-08-08:
Speichern ist das falsche Wort. Wird einfach nicht mehr aktualisiert.

closed 2025-08-22 by FireSon