Zendure/Zendure-HA #587: Devices not listed anymore after update to 1.1.0

Senfglasdichtung, 2025-08-08:
I have two Hubs (Hub 1200), both were integrated fine in the previous version (1.0.17?). Now only one is integrated, the one who is directly managed by my account. The other one managed by another account, but shared (Gerätefreigabe) with mine, is not.

After the update and reconfiguration of 1.1.0-pre19 the hub was still there with dead entities. Afer I reinstalled the integration the hub and its batteries were gone.

[home-assistant_zendure_ha_2025-08-08T08-36-51.738Z.log](https://github.com/user-attachments/files/21680173/home-assistant_zendure_ha_2025-08-08T08-36-51.738Z.log)

z-master42, 2025-08-08:
So funktioniert die Integration nicht (mehr). Es werden nur die Geräte angezeigt, die mit dem primären Account verbunden sind (da an diesen der Cloud-Token gebunden ist). Um den anderen Hub einzubinden ist der Cloud-Token von dessen Primärnutzer erforderlich und es wird ein eigenständiger Konfigurationseintrag.

FireSon, 2025-08-08:
@Senfglasdichtung The idea for using the token is that a secondary account is not needed anymore. So you should add all your devices to the same account, and use the token from that account. 
Is that possible, or do you have a special reason to keep using two accounts?

Senfglasdichtung, 2025-08-08:
No special reason, the second account is just not mine. It's a father-son thing.
I will put all devices into one account. But I think it would also be a good idea for the integration to support multiple accounts.

FireSon, 2025-08-08:
@Senfglasdichtung That is a valid point, I know of some other users who are in a similar position. Will think about a longer term solution for that!