Zendure/Zendure-HA #589: Question: Why is the MqttReset switch no longer available?

Christian23996, 2025-08-08:
Hello,

I currently have version 1.1.0-pre3 installed. The installation runs locally (mqtt local) with 3 Hyper2000. This version still has the MqttReset switch.

With this switch, I was able to "bring back" the Hyper2000 whenever it froze, which happened very frequently (at least 1-2 times a day).

This switch is no longer available in the current version.
Does this mean that it is no longer needed, i.e. the Hyper2000 no longer freezes?

Greetings, Christian

Christian23996, 2025-08-08:
Yes, that was my hope, stable Hyper2000!

Then I'll stick with version 1.1.0-pre3. With the MqttReset switch I can automatically reset the Hyper at any time.

FireSon, 2025-08-08:
Hi @Christian23996 ,

The reason the button is gone is basically, that in my opinion, the system should automatically restore the connection if it is lost. In the new version, if the connection is los for more than 5 minutes, it automatically tries to reconnect. You could say, that in that case the mqttRest button is automatically pushed. I have checked my 3 hypers as well, and I did not have a connection failure in the last 5 weeks.
There were some reports of 'freezes', but the ones I know of, and have investigated, had more to do with the MQTT server, because they were solved, once the MQTT server was restarted. We are monitoring this issue, and will try to improve dealing with connection failures!

@Geert2991 we did not reinvent the wheel, Zendure has released the zenSDK communication, which was incorporated into the integration. This meant restructuring quite a bit of code, but the actual communication with the Hyper2000 has not changed at all. 

Since you are missing it, I have added it back to the device, so in the next release it will be in 😄

closed 2025-08-08 by FireSon

FireSon, 2025-08-08:
@Geert2991 This reset at midnight was necessary, because as it turned out, the hyper had a memory issue, which was fixed in a master firmware release 2.1.30 In order to test this firmware, I removed the midnight reset, and tested my 3 hypers for multiple weeks, and I did not have any problem. 
There are indeed users, who expirience 'similar behaviour', but they can reset teir mqtt server, reconnect the hyper to the network which solves their problem. That was absolutely impossible with the original freeze. Therefore I think it is a different problem, so if you create a logfile when you encounter this behaviour, it would be much appriceated 😄

Christian23996, 2025-08-08:
Hey @FireSon,

Thanks for the explanation! That makes sense to me! 
I'll try out the new version for a while and test it.

Question: The new connection status only has four states. Could you please list them again with the corresponding numbers?

Greetings, Christian