Zendure/Zendure-HA #590: unique_id_mismatch

stdose82, 2025-08-08:
Hallo,
danke für die tolle Arbeit,
nach dem Update auf die Version 1.1.0, gebe ich die Daten wie beschrieben ein.
Den Token aus der App und bei P1 Sensor die Entität vom Shelly pro3Em.
Aber ich bekomme immer wieder den Fehler unique_id_mismatch.
Ich weiss gerade leider nicht weiter. Sorry, falls ich zu doof bin

Gruß Steffi

FireSon, 2025-08-08:
Hi Steffi, wo bekomst du diese Fehlermeldung? Hast du auch altere Versionen verwendet? Kannst du vielleicht ein screenshot machen?

stdose82, 2025-08-08:
Hi FireSon, das kommt direkt nach der Eingabe vom Tojen und des Senors.
Das steht im Logbuch

[image]

Logger: homeassistant.config_entries
Quelle: config_entries.py:749
Erstmals aufgetreten: 11:39:32 (1 Vorkommnis)
Zuletzt protokolliert: 11:39:32

Error setting up entry Zendure Integration - stdose@googlemail.com for zendure_ha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 749, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/zendure_ha/__init__.py", line 21, in async_setup_entry
    await manager.loadDevices()
  File "/config/custom_components/zendure_ha/manager.py", line 67, in loadDevices
    if self.config_entry is None or (data := await Api.Connect(self.hass, dict(self.config_entry.data))) is None:
                                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/zendure_ha/api.py", line 107, in Connect
    devices = await Api.ApiHA(hass, data)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/zendure_ha/api.py", line 129, in ApiHA
    raise ServiceValidationError(translation_domain=DOMAIN, translation_key="no_zendure_token")
homeassistant.exceptions.ServiceValidationError: Zendure token not provided

FireSon, 2025-08-08:
@flosch-dev @Kirschek and @flosch-dev 
This is probably due to changes in the configuration of the Integration. 
Could you try to delete the the entry here:

[image]

After that restart HA and add it again?

stdose82, 2025-08-08:
Hi FireSon,
it works, when you delete the old integration first.
Thanks

flosch-dev, 2025-08-08:
I can confirm, after deletion and re-creation (using the token) it works now.

harrymayr, 2025-08-08:
Is the token from the primary account? Please doublecheck if the token is complete.

5t0rmr1d3r, 2025-08-09:
> Is the token from the primary account? Please doublecheck if the token is complete.

No, the token is from the secondary account, who recieved the shared device.
I also checked the token again, it is correct.

5t0rmr1d3r, 2025-08-09:
> For this integration a secondary account is not needed anymore an the primary account.must be used.

Oh, didn't know that. That changed in the final 1.1.0
That can be more clearly described in the Chancelog (especially for users of <v1.1.0)

With the primary token everything worked, thx a lot!