Zendure/Zendure-HA #592: local mqtt not working with & in wifi password!

phlexss, 2025-08-08:
I have been trying for a long time to get local mqtt to work without success....also with 1.1.0 it didn't work.

Today I set up a new wifi access-point specifically for my Hyper2000 and assigned a new password to the wifi access-point without special characters and now local mqtt seems to work!

In my normal wifi password I have the character & and I now think that this was causing all the trouble; my Hyper2000 always went offline in the app when I switched to local mqtt and (so far) this seems to have been the reason.

Maybe the developers can do something with this info? Maybe fix the problem with &

I'll keep an eye on the hyper to see if it keeps working....

FireSon, 2025-08-08:
@phlexss Sorry to hear that. Would it be possible to send me a LogFile (with MqttLog on) when you are in local mqtt? Do you use Mosquitto? But can you also check your mqtt broker, to see if it logs any error?

phlexss, 2025-08-08:
Yeah......rooky mistake.....just copied it from another integration without thinking.....that seemed to have done the trick.

Feeling a little stupid now but glad it works!