Zendure/Zendure-HA #595: What is expected in this configuration screen for local mqtt

surfer1264, 2025-08-08:
Hey team 
SF1200/AB1000S/AB2000

I installed 1.1.0 with Cloud-Key
worked well (big applause)

I would like to try the local mqtt part,
I red 

https://github.com/Zendure/Zendure-HA/wiki/Installation

but what is expected (maybe some silly question) for 

- mqttserver (I guess the IP-address of my HA Instance, is this right?)
- mqttuser
- mqttpassw

[image]

I run a HA Instance with Mosquitto Mqtt Broker and manual configured Zendure Device (@z-master42 approach) on HA Green

[image]

Both instances zmaster42 and Zendure-HA are working currently in parallel (this is not a problem until now)

Thanks for some advice

harrymayr, 2025-08-08:
> Ich habe im Mosquitto keinen User angelegt.

Ich hätte jetzt erwarte, der wird automatisch angelegt? Ist aber schon eine Weile her, dass ich ihn installiert habe:

[image]

surfer1264, 2025-08-09:
Also was ist die finale Antwort?

MQTTuser  = HA User
Mqttpw = das zugehörige PW

Und wieso brauche ich das überhaupt, wenn ich im HA Kontext unterwegs bin und mosquitto von jedem HA User benutzbar ist.

So ganz richtig verstanden habe ich das ( leider) nicht.

Interessant war der Post eines Users im Zendure Forum
Im Zendure Forum schreibt gerade jemand...(TL-F): Name und Passwort der Zendure App und Juhu es funktioniert

Klingt so als wäre es völlig egal was man da angibt.

z-master42, 2025-08-09:
Auch im HA Kontext muss man sich am Broker anmelden, also muss man da auch Anmeldedaten angeben.

Was nicht so ganz klar ist, ist ob die Integration jetzt einen Benutzer erstellt oder nicht, weil so liest sich die veraltete Doku.

Was kam beim Probieren denn jetzt raus?

closed 2025-08-10 by harrymayr