Zendure/Zendure-HA #606: Feature Request: Manual Mode +

Popoboxxo, 2025-08-09:
First off, I want to say that the new update is running super smoothly! The integration with the Zendure system, especially with the EOS/EOS Connect features, is fantastic. It's not just about achieving "zero export", the new modes like Smart Load are working perfectly!

**Requested Feature: "Manual +" Mode**
I'd like to request a new feature to address a specific corner case. Currently, if I set the system to Manual Mode to charge the batteries with a predicted AC load (e.g., 400 W), and an unexpected increase in solar power occurs (e.g., 1400 W), the system might begin exporting power to the grid. For instance, if the base load of the house is only 300 W, the system will export the remaining 700 W.

I've temporarily solved this with an automation that switches back to Smart Load mode whenever the system starts exporting more than 30 W while in Manual mode.

To improve this, I propose a new mode, which could be named "Manual +," "Manual - Offset," or "Manual - Smart Load." This mode would operate similarly to the current Manual mode but with an added feature: it would constantly monitor for grid export. If it detects that the system is exporting power, it would automatically increase the AC charging power to use up the surplus solar energy, preventing the export from happening. This would allow users to maintain their desired manual charging rate while still intelligently responding to unexpected increases in solar production.

harrymayr, 2025-08-09:
> First off, I want to say that the new update is running super smoothly! The integration with the Zendure system, especially with the EOS/EOS Connect features, is fantastic. It's not just about [...]

Basically it's a smart matching with and charging offset. I had in one of the previous versions implemented an offset of +/- 50W, so that it trys not to reach 0W, but the offset you entered.
There is a plan to integrate that in the 1.1.x, but for now we had not considered that high amount, But we can think of +500W (meaning charging) and -50W.

FireSon, 2025-08-14:
@Popoboxxo Thanks for the proposal! In this 'area' a few requests were made, and once 1.1.x is stable, we will look into it 😄