Zendure/Zendure-HA #608: Can't set it up

wagnerpizza, 2025-08-09:
Using 1.1.0 pre19 -> What shows the current release version 1.1.0 in home assistant

I followed the instructions with Token and MQTT settings but get these error

[image]

Version with 1.0.47 runs for months without problems.

harrymayr, 2025-08-09:
Delete Integration, restart HA and add it again

wagnerpizza, 2025-08-09:
@harrymayr 
Deleted -> Restart -> new Installation (without local mqtt and an not existing Smart Matching Sensor) -> same result

[image]

harrymayr, 2025-08-09:
> BTW the token is from second account like the suggestion of previous version

That's the problem. 2nd account will not work an is not needed anymore.

harrymayr, 2025-08-09:
For the smart matching. You can put in your shelly (sum of all 3 phases).

wagnerpizza, 2025-08-09:
@harrymayr 
Ok got it running. But maybe some stupid questions, where to activate the "smart Matching"? 
How to check the values of local MQTT? I run MQTT Explorer and cant see any topic from zendure or simillar. 
"Local means" it will work without internet connection as well right? 
If I turn on "local" can I still access via Zendure App?
When I acitvate the Zendure Manager, it will overrule the cloud connection?

closed 2025-08-10 by harrymayr