Zendure/Zendure-HA #615: Hyper Freeze (again)

Kieft-C, 2025-08-10:
V1.1.1-pre2 in HA
Mit v2.1.30 im Hyper Firmware

Connection Status wechselte von 1 auf 0 und mit addon mosquitto Neustart ging es wieder.

Identische Problematik wie bei 1.1.0-pre3. Hing sich auf und via automation hab ich dann addon neu starten lassen.

Sollte ja mit Update bei Zendure gelöst werden, wenn ich mich richtig erinnere ?
Kann es ggf an Generationen von Hardware liegen?

Kieft-C, 2025-08-11:
[image]

[image]

Heute Morgen 03:30 wieder Freeze vom mqtt und Neustart des addons hat es gefixt.
Das Protokoll ist leider so lang das ich es am handy nicht einsehen kann. Muss es am pc herunterladen und zur Zeit scrollen nachher.
Generell sind da echt ne Menge Daten im log. Irre.

Kann man dies zwischenzeitlich mal löschen? Cache mäßig?

FireSon, 2025-08-14:
@Kieft-C The information from the Mosquitto log might be interesting, this is definitly a 'different' kind of freeze. It is worth a try to reset the mqtt connection when this problem occurs. When we would add a button like the mqttReset, which would 'reconnect' to Mosquitto, would you be able to test it?

Linus86529, 2025-08-14:
Please excuse me for shaking my head because there is something I absolutely do not understand and cannot comprehend. Why is such an effective and proven tool as the MQTT reset button being removed from all versions after v1.1.0-pre3? MQTT is still the key element in integration. 

The MQTT reset button is also a reason why I keep reinstalling v1.1.0-pre3. In the pre3 version, I don't have to restart the entire MQTT integration when I encounter MQTT problems with the Z-HA integration.

The MQTT reset should only be removed from the code when it is no longer necessary.

Kieft-C, 2025-08-15:
@Linus86529 
The reason behind is, that with a update for the hyper the "old" freeze situation should be gone.
The new freeze could something else. so the need for the button was not neseccary anymore and the other new freeze we try to find out where it cames from.

Kieft-C, 2025-08-16:
got a freeze right now. automation started addon mosquitto new.
Timestamp see at screenshot.

[core_mosquitto_2025-08-16T10-53-36.019Z.log](https://github.com/user-attachments/files/21811973/core_mosquitto_2025-08-16T10-53-36.019Z.log)

[image]

FireSon, 2025-08-19:
@Kieft-C If I look at your logfile, it seems that the 'same' client sends a PingReq each second. This is not expected, since the pingreq is used to keep the connection 'alive', by sending such a ping every 30 seconds or so. 

```
2025-08-16 12:45:18: Received PINGREQ from KTEfLQ5L
2025-08-16 12:45:18: Sending PINGRESP to KTEfLQ5L
2025-08-16 12:45:19: Received PINGREQ from KTEfLQ5L
2025-08-16 12:45:19: Sending PINGRESP to KTEfLQ5L
2025-08-16 12:45:19: Received PINGREQ from KTEfLQ5L
2025-08-16 12:45:19: Sending PINGRESP to KTEfLQ5L
2025-08-16 12:45:20: Received PINGREQ from KTEfLQ5L
2025-08-16 12:45:20: Sending PINGRESP to KTEfLQ5L
```
One possibility is that there are multiple connections open. Can you reboot your HA machine, and then create another Mosquitto logfile? Just to see if these logentries are different.

Kieft-C, 2025-08-24:
@FireSon any chance to make a step here? :) how I could help?

pulsarz, 2025-08-26:
Haven't had a freeze for like a week and then yesterday got one (on the new revision hyper..) and today in my other one.

Each time restarting EMQX fixed the issue.

Guess I'll have to make a nodered flow to restart the docker container but this is really not great..

pulsarz, 2025-09-10:
> [@Kieft-C](https://github.com/Kieft-C) Sorry I was busy with a new version 😄 I did try to look into the error you have. But it is very difficult to find a solution. I did find one similar issue on the internet, and put their solution into version 1.1.4-pre3 Not really sure that it will help though... Can yue test it? Please (re)open the issue if there is still a problem!

Sadly it is not solved, had yet another freeze. My nodered workflow detected it and restarted EMQX and it works again.

So.. no change.