Zendure/Zendure-HA #616: Disabled entries that have been removed in (1.1.0) update can't be removed

noseshimself, 2025-08-10:
I have disabled several sensors because they were not relevant to my system:

[image]

In order to get fully rid of them I tried enabling them as disabled entries can't be deleted. 

Turns out: You can't delete them when enabled either because they do not receive a value and thus

[image]

with the "Delete" button being disabled.

I don't know if this has to be addressed by the core providing that function or if it is a problem with this integration but even if it is a core problem it would be nice if you forwarded this to the correct project.

harrymayr, 2025-08-10:
Even on a restart of HA, these sensors are still visible or not able to delete?

Linus86529, 2025-08-31:
I have a similar problem with the SF800, which was deliberately disabled because the home-assistant.log was full. After restarting the HA system, a second SF800 was detected. The screenshot shows the ghost SF800 with fewer entities than the original SF800.

By the way, a strange observation from today. The problem with the ghost batteries was fixed a few weeks ago. I have no idea why or how this could happen again. The screenshot shows two ghost batteries that have fewer entities than the original batteries.

I think it's time for a clean install again.

[image]

Linus86529, 2025-08-31:
I forgot to mention that the value of 'Verfügbare Energie' was calculated correctly after deactivation of the two Ghost Batteries and after pressing 'RESET Verbindung' for the Hub1200 and theHyper2000.

noseshimself, 2025-10-01:
[image]

[image]

The problem is still there and I still believe that it is a general Home-Assassin-UI problem.  If it is enabled, all I can do is disable it. If it is disabled, all I can do is enable it. There just is no UI element to delete it at all...