Zendure/Zendure-HA #618: 1.1.1: Smart Matching no longer working

d0g2342, 2025-08-10:
After updating from version 1.1.0 to version 1.1.1, the 'smart matching' feature stopped working.
SolarFlow_800_Output_Limit stays at 0; if I change the value manually, it works, but it seems that the automation expected from 'smart matching' is broken after the update.

I'm using an SF800 with zenSDK as the connection type.

Please feel free to ask for any further information you need.

itapower, 2025-08-10:
Same issue here on my Hyper 2000 after updating smart matching not working

d0g2342, 2025-08-10:
> > After applying the pull request from [@harrymayr](https://github.com/harrymayr) it works again ([#617](https://github.com/Zendure/Zendure-HA/pull/617))
> 
> Hmm... what does this mean? What to do?

wait for a new release or modify line 135 in "custom_components/zendure_ha/manager.py":
self.update_p1meter(data.get(CONF_P1METER, "sensor.power_actual"))
should be
self.update_p1meter(self.config_entry.data.get(CONF_P1METER, "sensor.power_actual"))

DerUnwissende, 2025-08-10:
Ok... thanks @d0g2342 ... Changed the line by myself... 😄

d0g2342, 2025-08-10:
Solved in Version 1.1.2

closed 2025-08-10 by d0g2342