Zendure/Zendure-HA #630: 1.1.2-pre2: multiple entities of SolarFlow 800 Pro are unavailable

tmhrzgr, 2025-08-10:
With 1.1.2-pre2 multiple entities of the SolarFlow 800 Pro-device are suddenly unavailable. In 1.1.0 everything worked fine. Reloading the integration, new download of the whole HACS-integration and waiting (for a long time :)) won't solve the issue. Please have a look, thank you! #

[image]
[image]

phlexss, 2025-08-11:
I had the same problem with my Hyper2000....al lot of entities were missing/unavailable....going back to 1.1.0 fixed the problem.

z-master42, 2025-08-11:
Have you tried restarting the Home Assistant a second time?

What happens to the entities when you go to the battery level overview page in the Zendure app, where you can also see the battery temperatures (assuming the mode is cloud and not local)?

harrymayr, 2025-08-11:
please try out 1.1.3

harrymayr, 2025-08-12:
I see this error also, it looks like, that it will try to setup too early in the initialization phase. But later it set the fusegroups with no eerror. Using Smart mode normally needs to set a fusegroup, maybe your disabled Hub mix this up. Does it show up as a 2nd device?

tmhrzgr, 2025-08-12:
> please try out 1.1.3

No issues with SF800Pro anymore, all entites are available. Thank you! :)

closed 2025-08-12 by tmhrzgr