Zendure/Zendure-HA #632: Local mode: Does the integration talk directly with zendure cloud!?

phlexss, 2025-08-11:
Today (using 1.1.0) I disabled internet access of my Hyper2000....and to my astonishment I kept getting updates in the app!? I'm running in local mode.

This must mean the HA integration is directly talking to the cloud even when in local mode.

For me the whole point of running in local mode is that I don't want to be connected to the cloud or update the cloud with what my hyper is doing.....but now it seems I can't block this at all. I thought blocking internet access of the hyper would block the app (and zendure) from getting any updates from my hyper but now this doesn't seem to be the case.

Ofcourse blocking internet access of my Home Assistant machine is no option.

Is it possible to add a switch to the integration to disable the updates to the cloud when in local mode? ....I don't need the app and I don't want zendure to get any updates when in local mode.

phlexss, 2025-08-11:
It seems if I block domain mqtteu.zen-iot.com in my adguard the app doesn't receive updates anymore and so far the integration keeps working just fine..... :-)

But a switch to turn this off or on in the integration would be great!

z-master42, 2025-08-11:
This is how it works at the moment.

https://github.com/Zendure/Zendure-HA/wiki/Local-Mqtt#zendure-app

As I understand it, @harrymayr is planning to install a switch that can be used to switch this off.

phlexss, 2025-08-11:
> As I understand it, [@harrymayr](https://github.com/harrymayr) is planning to install a switch that can be used to switch this off.

That would be awesome!

harrymayr, 2025-08-11:
> > As I understand it, [@harrymayr](https://github.com/harrymayr) is planning to install a switch that can be used to switch this off.
> 
> That would be awesome!

Not yet a switch, but if you delete this 8 lines in api.py, there should be no sendout of messages to the Zendure Server. In this case your device should show Offline in the App, while you get data in HA
There is also a new version 1.1.3 you should install (and from where this code snippet is). This fixes the issue if minSoc or maxSoc are reached.

[image]

phlexss, 2025-08-12:
@harrymayr 
I installed 1.1.3 and commented these lines....so far it looks good: no more updates in the app and (so far) no strange behaviour or excessive mosquitto logging.

Thanks! Will there be a switch for this in the future?

So far the hyper doesn't go offline in the app but I noticed the same when blocking the domain...it just shows the last values of when it got the last update.....but maybe it'll go offline after some timeout.....

z-master42, 2025-08-12:
As far as I know, updates only work via the cloud, apart from that I don't assume that the Hub 1200 will get any updates.

The fact that inverse_max_power cannot be changed via integrity is a deliberate decision (I was told so).

Linus86529, 2025-08-12:
Hi @z-master42, sorry, I forgot to mention that I have a Hyper2000 and an SF800. The Hub1200 will be replaced by the SF800 next month. And these boxes will still get various updates.

phlexss, 2025-08-15:
3 days later and the Hyper still is not offline in the app but still shows the same values as 3 days ago.

Not a huge problem and I chose to do this but now it's also not possible to connect the Hyper over bluetooth....

The weird thing is that I restarted my wifi during this period so you would expect the hyper to go offline because the cloud couldn't get any recent updates but it doesn't seem to do that.....

phlexss, 2025-08-22:
Too bad that 1.1.4-pre2 doesn't contain a switch yet to enable/disable the update to the cloud.

I have made my own "solution" for now:

1. in const.py add:
`CONF_UPDATE_CLOUD = "0"
`
...or set to 1 to enable the update to the cloud.

2. in api.py add:
`if CONF_UPDATE_CLOUD == "1":`
before:
`                        if device.zendure is None:
                            psw = hashlib.md5(device.deviceId.encode()).hexdigest().upper()[8:24]  # noqa: S324
                            device.zendure = mqtt_client.Client(mqtt_enums.CallbackAPIVersion.VERSION2, device.deviceId, False, "zendure")
                            self.mqttInit(device.zendure, Api.cloudServer, Api.cloudPort, device.deviceId, psw)

                        if device.zendure is not None and device.zendure.is_connected():
                            payload["isHA"] = True
                            device.zendure.publish(msg.topic, json.dumps(payload, default=lambda o: o.__dict__))`

So it will be:

`                    if CONF_UPDATE_CLOUD == "1":
                        if device.zendure is None:
                            psw = hashlib.md5(device.deviceId.encode()).hexdigest().upper()[8:24]  # noqa: S324
                            device.zendure = mqtt_client.Client(mqtt_enums.CallbackAPIVersion.VERSION2, device.deviceId, False, "zendure")
                            self.mqttInit(device.zendure, Api.cloudServer, Api.cloudPort, device.deviceId, psw)

                        if device.zendure is not None and device.zendure.is_connected():
                            payload["isHA"] = True
                            device.zendure.publish(msg.topic, json.dumps(payload, default=lambda o: o.__dict__))`

And restart HA...also restart HA if you change the value of CONF_UPDATE_CLOUD 

....I still hope they add a switch for it because I couldn't figure out how to do this myself.

FireSon, 2025-08-22:
@phlexss The change to do it is not very difficult.  But I am waiting for a response from Zendure about 'disconnecting' from the cloud. Just want o be sure what possible consequences can be. 
For instance:
* void of warrenty
* scheduling calibration 
* check for firmware updates
* collecting device logging

FireSon, 2025-08-22:
Apperantly this is a sensitive subject 🥴 

I just want to know what Zedure is doing with the data, and if there is a possible downside to 'disconnecting'.
Nothing more. Please stop reading between my lines, I have been transparant about this all the time. 

So relax ☺️ and I will come back to this topic, once I have the information.

phlexss, 2025-08-22:
> > Apperantly this is a sensitive subject 🥴
> > I just want to know what Zedure is doing with the data, and if there is a possible downside to 'disconnecting'. Nothing more. Please stop reading between my lines, I have been transparant about this all the time.
> > So relax ☺️ and I will come back to this topic, once I have the information.
> 
> Sorry [@FireSon](https://github.com/FireSon) but then don't give these strange examples of why disconnecting might be a problem.....I too want to know if there are downsides but none of your "for instances" makes sense.

And yes, data harvesting and cloud dependancy are a sensitive subject.

FireSon, 2025-08-22:
@phlexss why do you have to be the judge of what is strange, man this is really getting annoying!!

MacdiverOne, 2025-08-22:
Just a friendly reminder that this important discussion should remain objective and substantive at all times and should not descend into politicization or personal assumptions about the values of the developers, whom we are grateful to for their invaluable (and free!) work. Well-founded concerns and arguments have been clearly and, for the most part, objectively presented and, as far as I understand, will be taken into account as quickly as possible.

Just my 2ct.

phlexss, 2025-08-26:
It seems that, even with the cloud-update code disabled,  the integration still connects with mqtteu.zen-iot.com:

I noticed this in my AdGuard logging.

I have now added a rewrite rule to AdGuard to redirect mqtteu.zen-iot.com to my HA ip.....then I see connects coming in in my mosquitto logging from a new client with the ip address from my Home Assistant box (so it's coming from the integration)....and this connection is denied because it's from an unknown user......this happens constantly every two minutes.

Functionality doesn't seem affected though....just the added logging in mosquitto and I see the rewrites every two minutes in the logging of AdGuard.

The above is not to annoy anybody but I thought the cloud would be disconnected with the code disabled as described earlier in this post. However, the integration still tries to talk to the cloud and I'd like to understand why that is, what it is for and if it can be disabled.

Any info on this is appreciated.

closed 2025-10-02 by github-actions[bot]