Zendure/Zendure-HA #633: Hyper 2000 - Unplausible Werte ggf. fehlende Beschreibung

GrisuDerDrache, 2025-08-11:
Hallo liebe Entwickler/Programmierer,

folgende Sensoren zeigen bei mir unerklärliche Werte an. Ich habe mal versucht an Hand der Datenpunkt-Beschreibungen mir eine Erklärung zu recht zu legen. Wenn ich falsche liegen sollte mit meinen Vermutungen und Halbwissen, dann bitte mich schnell verbessern. Falsche Interpretationen bitte sofort löschen, sonst führen diese nur zur weiteren Verwirrung bei anderen Nutzern.

* sensor.hyper_2000_energy_power -> "keine Übersetzung / Beschreibung hinterleget" -> zeigt einen dauerhaft wert z.B. 1200 W an -> vielleicht fehlt auch nur die richtige Übersetzung

[image]

* sensor.hyper_2000_pv_brand -> "Wechselrichtermarke" ( in dem Wert kommt wahrscheinlich nur 0 oder 1 vor -> vermutlich Erkennung ob das interne Brandlöschsystem  ausgelöst hat?? ggf. falsche Übersetzung oder ich bin auf dem Holzweg mit meiner Vermutung)

[image]

* sensor.hyper_2000_volt_wakeup -> "keine Übersetzung / Beschreibung hinterleget" ->  zeigt dauerhaft den "Wert" Null an. Ist das vielleicht ein Überwachungswert, ab einem unterschreiten eines bestimmten Spannungswert von einer Akkuzelle, das ein tiefentladen der Zelle verhindert werden soll? Richtig oder falsch?

[image]

* sensor.hyper_2000_switch_count -> "keine Übersetzung / Beschreibung hinterleget" -> Wert schaltet ja nach Zustand bei mir zwischen den Werten 1-11 . Was soll der Wert uns zeigen?

[image]

* sensor.hyper_2000_available_kwh -> "Verfügbare Energie" -> Werte "springen" zum Teil, das beobachte ich weiter
   vermutlich kommen die Sprünge zustande wenn die Integration neu gestartet wird.

[image]

* binary_sensor.hyper_2000_reverse_state - > "keine Übersetzung / Beschreibung hinterleget" -> Was kann damit gemeint sein?

[image]

* sensor.hyper_2000_masterhaer_version -> "Hardware-Version" -> wird bei meinen Hyper nicht ausgelesen, Antwort des Datenpunktes "not provided"

[image]

z-master42, 2025-08-11:
Ja, einige Übersetzungen fehlen noch, bzw. ist die Bedeutung dessen was die Entität anzeigt noch nicht überall klar, da die Integration alles anzeigt, was sie geliefert bekommt.

Da die originalen Bezeichnung aber alle Englisch sind steht Brand für Marke. Die Entität ist beim Hyper aber tatsächlich sinnlos, bzw. könnte man die 0 als integriert verstehen.. Bei den Hubs repräsentiert die angezeigte Ziffer die in der App ausgewählte Wechselrichtermarke.

GrisuDerDrache, 2025-08-17:
Siehe hier ...

https://github.com/Zendure/Zendure-HA/discussions/658

closed 2025-08-17 by GrisuDerDrache