Zendure/Zendure-HA #634: Version 1.1.2pre2 , zendure manager not work again...

pacostef57, 2025-08-11:
Last night, the smart matching mode was working. At midnight, I switched it to Off mode to switch it back to smart matching this morning with solar production. This morning, despite excess power to the grid, smart matching does not start charging the battery. In the Zendure app, I can see 'the energy plan' in the green banner at the top, but nothing more.

d0g2342, 2025-08-11:
Same here. HA was on 'smart matching' yesterday, which worked, but this morning no more regulation took place. After I switched from 'smart matching' to 'smart charge', the battery started charging (it was at 17%). I then switched back to 'smart matching' and everything seems to be working again.

harrymayr, 2025-08-11:
There are also Energy plans for the device itself. These where used, before HEMS take place.

z-master42, 2025-08-11:
Yes, HEMS or, if you don't use HEMS, the energy plans in the app.

d0g2342, 2025-08-11:
> Yes, HEMS or, if you don't use HEMS, the energy plans in the app.

Ok. That's what I did. At least for the SF800 there is no possibility in the app to turn the energy plans completely off.

pacostef57, 2025-08-11:
> I will check all the logs this evening and try to find the issue with switching from charge to discharge. It looks like, it has to do with the newer devices which use ZenSDK or with more than 1 devices. For my single Hyper I don't see any problems.

Aber ich habe nur einen Hyper2000, der direkt an die Steckdose angeschlossen ist, und trotzdem hat der Smart-Matching-Modus heute Morgen nicht gestartet... er startete erst, als ich ein wenig mit den Modi gespielt habe und den Akku aus seinem minimalen Zustand genommen und dann in den lokalen Modus gewechselt bin, also weiß ich nicht wirklich, was das Problem gelöst hat.

CommanderROR9, 2025-08-11:
I think there is some kind of "freeze" happening. After a certain time my SF2400 appears to just stop accepting commands. This leads to the fact that it then doesn't switch from Charge to Discharge. It then also doesn't appear to accept the "manual" commands from the Zendure Manager anymore. So maybe the "Manager" part is somehow getting stuck or the API is getting overwhelmed somehow?

pacostef57, 2025-08-11:
See #641 , Once the smart matching started this morning, it worked well until it fully charged the battery. Now that I need it to discharge, it remains stuck at 0.

pacostef57, 2025-08-11:
> Cela ressemble au problème habituel. Je me demande si je pourrais créer une sorte d'automatisation pour le faire passer en mode Décharge… comme solution de contournement pour l'instant… Je vais y réfléchir 🤔

I have no doubt they will debug this very quickly; it must be a blockage in a mode. I am unable to help more than providing logs; I am not a programmer. It seems like not much is missing for it to be okay.

pacostef57, 2025-08-11:
> I will check all the logs this evening and try to find the issue with switching from charge to discharge. It looks like, it has to do with the newer devices which use ZenSDK or with more than 1 devices. For my single Hyper I don't see any problems.

English:
"I have certainly just found the problem. I discharged my battery a bit, which was at 100%, in manual output mode (hyper2000) just so that socLimit would display 0. Then, I stopped the manual discharge in hyper, and I went into Zendure Manager to activate the smart matching. Then, after 30 seconds, it started properly and is now regulating the discharge again."

German:
"Ich habe sicherlich gerade das Problem gefunden. Ich habe meine Batterie, die bei 100% war, ein wenig im manuellen Ausgangsmodus (hyper2000) entladen, nur damit socLimit 0 anzeigt. Danach habe Ich die manuelle Entladung in hyper gestoppt und bin in den Zendure Manager gegangen, um das Smart Matching zu aktivieren. Dann, nach 30 Sekunden, hat es richtig gestartet und reguliert jetzt wieder die Entladung."

CommanderROR9, 2025-08-11:
Yes. But I think the question is, whether the SoC.is the issue, or the fact that ot uas entered some kind of rest mode and you just "jiggled" it out of that.

closed 2025-08-12 by pacostef57