Zendure/Zendure-HA #639: Failed to connect to Zendure API

Raptor8604, 2025-08-11:
After updating to 1.1.1 i am unable to connect to zendure cloud.
Debuglog:
2025-08-11 14:14:33.409 ERROR (MainThread) [custom_components.zendure_ha.api] Failed to connect to Zendure API

Error aufter submitting Cloud-Token: unique_id_mismatch

Nadine991, 2025-08-11:
Bei mir auch.

morgeek, 2025-08-11:
same here My entities are not visible anymore

harrymayr, 2025-08-11:
unique_id_mismatch may be happen, if you use the 2nd account. You need to use the 1st account for the integration. 2nd account is not needed anymore.
Also comming from 1.0.47, you need to delete the integration, restart HA and install the integration again.
Also a new version 1.1.3 is out which you should use.

closed 2025-08-14 by FireSon