Zendure/Zendure-HA #645: Fehlermeldung - Gerät durch Konfigurationseinstellung deaktiviert

mangri01, 2025-08-11:
Ich habe heute die Version 1.1.2 HACS Installiert. Vorher alles Deinstalliert und HA Restart gemacht.
Alles Eingerichtet
Token vom ersten Zendure Account aus der App eingetragen
sensor.power_actual per Helfer angelgt. Zeigt positiv und negativ an wie der shelly 3em pro

Wieder alles Deinstalliert, paar mal HA restart.
Version 1.1.1 Installiert alles eingerichtet wieder gleiches Problem wie oben.
Wieder alles Deinstalliert, paar mal HA restart.
Version 1.0.47 Installiert alles eingerichtet wieder gleiches Problem wie oben.

Mein System 
HA 2025.8.0

4 x SolarFlow 1200 (Firmware SF und AB Aktuell) an jedem SF hängt ein AB
2 x Akku AB1000
2 x Akku AB2000
Shelly 3em pro per LAN
2 x HMS 1800

[image]

[image]

mangri01, 2025-08-11:
Durch einen Tipp die Integration **Deaktivieren und wieder Aktivieren**. Und die Garäte waren nicht mehr ausgegraut.
Jetzt werde alle Entitäten wieder angezeigt.

mangri01, 2025-08-11:
Nun habe ich im Gerät bei den Steuerelementen nicht mehr alle Entitäten. Einige sind Deaktiviert, ich weiß aber wie man Sie Aktivieren kann. Oder soll man die Entitäten nicht Aktivieren.

**Nun meine Frage. Was soll man bei der Entität Sichertsgruppe und bei Verbindungsart einstellen.** Natürlich kann man da nur etwas einstellen, wenn ich die Entitäten Aktiviere. 

[image]
[image]

surfer1264, 2025-08-11:
a) WARTEN und WARTEN 
Die Entitäten kommen LANGSAM rein

b) mind. eine Sicherungsgruppe auswählen, wenn Du die smarte Steuerungsfunktion nutzen willst, ansonsten nicht relevant

mangri01, 2025-08-11:
Stimmt das so wie ich die Sicherheitsgruppe eingestellt habe. Nutze ha 4 x Solar Flow 1200 und an jedem Solar Flow hängt eine AB Akku

[image] SolarFlow 1

[image] SolarFlow 2

[image] SolarFlow 3

[image] SolarFlow 4

mangri01, 2025-08-11:
Ich verstehe nicht, weshalb mein System nicht ordentlich regelt.

surfer1264, 2025-08-11:
Debuglog anwerfen und den Devs hier zur Verfügung stellen.

mangri01, 2025-08-12:
Muss man in der Zendure App etwas einstellen.

Momentan habe ich keinen Energieplan ausgewählt, die Ausgangsleistung steht bei 900 watt.

surfer1264, 2025-08-12:
Nein eigentlich nicht.

Mein SF1200 läuft im Basismode
Ausgangsleistung voll aufgedreht.
Kein Energiemodus gewählt

Du hast eine Sicherungsgruppe gewählt mit 3600 W, bedeutet alle 4 SF1200/ hängen auf der gleichen Phase hinter einer Sicherung ( das bedeutet Fuse Group, laut Beschreibung)

(Für klassische BKWs wäre das nicht konform, es sei denn so angemeldet und abgenommen)

Die smarte Steuerung sollte nun über den P1 Sensor die richtige Energiemenge einsteuern, so mein Verständnis.

mangri01, 2025-08-12:
Hallo Surfer,

ja bei meine SolarFlow habe ich keinen Energieplan, Ausgangsleistung bei allen vier 900watt.

**Aber ob die Sicherungsgruppe richtig ist weiß ich nicht???**

Drei SolarFlow hängen an einem Kabel und geht in den Keller zum Sicherungsschrank
Einer hat ein extra Kabel in den Keller zum Sicherungsschrank
wo der Shelly 3em pro steckt

Was muss man bei Verbindungsart HA Zendure einstelle, siehe Bild.

[image]

Wenn ich mit diesen Einstellungen die Smarte Steuerung starte regelt das System nicht.

surfer1264, 2025-08-12:
Wie gesagt debugfile produzieren und hier einstellen, ich kann hier nicht mehr weiterhelfen. Ich habe dein Setup nicht. Müssen sich die devs hier anschauen.

Warum deine 4 SF1200 mehr einspeisen al sie sollten.
@harrymayr

mangri01, 2025-08-12:
Wie starte ich das Debugfile

harrymayr, 2025-08-12:
Ich schaue.mir das heute Abend mal an.

harrymayr, 2025-08-12:
OK, erst mal ein Hinweis, der nichts mit der Integration zu tun hat. Du hast jede Menge Fehler im Log, die damit zu tun habe, dass bestimmte Werte (noch) unknown sind (siehe unten).
Die kannst du verhindern, indem du dem `float` einen  Standardwert gibst, z.B.

`{{ (states('sensor.solar_einspeisung_pv_module') | float(0)) - (states('sensor.pv_einspeisung_alle') | float(0))}}`

[image]

harrymayr, 2025-08-14:
Hallo Manfred,

ich werde leider nicht richtig schlau daraus :-| 
Hast du es mal im Cloud-Mode versucht?

Was noch eine Möglichkeit wäre um dem Problem auf die Spur zu kommen:
- Debug logging ein, 
- Zendure Manager auf Aus
- Verbindungsart: Cloud
- HA neu starten
- Wenn die Entitäten alle da sind, in der App entweder über HEMS mit deinem 3EM steuern, oder aber Smart CT Energieplan mit dem 3EM starten.
- nach 5-10 Minuten debug-logging ausmachen
- Dann wieder das einstellen, was bei dir funktioniert.

Mein Ziel ist es zu prüfen, wie die Daten aussehen, wenn die Zendure App das alles steuert.

mangri01, 2025-08-15:
> Hallo Manfred,
>
> ich werde leider nicht richtig schlau daraus :-| Hast du es mal im Cloud-Mode versucht?
>
> Was noch eine Möglichkeit wäre um dem Problem auf die Spur zu kommen:
>
>     * Debug logging [...]

ich habe alles nach deiner Anleitung gemacht und einen Hub SolarFlow1 in der App mit dem shelly 3em verbunen und den Smarten Ableichsmodus eingestellt. 
Der Hub regelt nicht, dachte es liegt villeicht an der Ausgangsleistung, die bei allen vier auf 0 gestellt. Smarter Modus in der App erneut bei SolarFlow 1 gestartet.
Hub SF1 regelt nicht

Normal steuere ich meine 4 SF und meine beiden HMS 1800 mit einer openDTU on Battery. Das läuft auch so wie es soll.

Anbei das Logfile

[home-assistant_zendure_ha_2025-08-15T05-05-15.225Z.log](https://github.com/user-attachments/files/21789706/home-assistant_zendure_ha_2025-08-15T05-05-15.225Z.log)

harrymayr, 2025-08-15:
Ok, wenn der Hub mit der App nicht regelt, dann liegt das wahrscheinlich daran, dass die Umschaltung auf Cloud noch nicht erfolgt war.
Ich habe mit meinem Hyper Probleme mit der Umschaltung, da die BT Verbindung zwischen HA und dem Hyper sehr oft nicht stabil ist.
Ggf war das bei dir auch der Fall.

Mit OpenDTU on Battery habe ich keine Erfahrung, eine kurze Suche hat aber ergeben, das dafür ggf. ein Energie Programm `Appointment mode (08:00 800W/20:00 150W)` eingestellt werden muss/kann.

Hast du auch sowas noch an, wenn du den Smart Mode testest?

Log schau ich mir heute Abend an, wenn allerdings nichts geregelt wurde, werde ich da nicht viel Info rausgehen.

harrymayr, 2025-08-15:
Im MQTT Explorer findest du aber Einträge wie hier im [Post](https://github.com/Zendure/Zendure-HA/discussions/638#discussioncomment-14097862), allerdings mit anderen Einträgen. Deine 4 Hubs müssten da zu sehen sein, wenn du im Lokalen Mode bist.

harrymayr, 2025-08-15:
Aber was macht openDTU in dem Moment, der regelt ja den Wechselrichter (nach meinem Verständnis)

mangri01, 2025-08-17:
Ja die opdenDTU regelt den Wechselrichter.
Wenn ich die Steuerung SolarFlow im HA ausprobiere schalte ich die openDTU aus.

Ich verstehe nicht, weshalb meine 4 SolarFlow nicht geregelt werden von der Zendure Home Assistant Integration.

Ich habe scho so viel ausprobiert geht nicht.

DerUnwissende, 2025-08-17:
> Ja die opdenDTU regelt den Wechselrichter. Wenn ich die Steuerung SolarFlow im HA ausprobiere schalte ich die openDTU aus.
> 
> Ich verstehe nicht, weshalb meine 4 SolarFlow nicht geregelt werden von der Zendure Home Assistant Integration.
> 
> Ich habe scho so viel ausprobiert geht nicht.

hmmm... schaltest du opendtu aus, weil du am probieren bist? Ich habe die immer an... Und überhaupt, ziehst du den Stecker der opendtu? 
Hast du schon mal versucht nur 1 Solarflow anzuregeln? die restlichen erstmal nicht zu betrachten? Wenn das dann gehen sollte, dann die Nr. 2 hinzu usw. ?

mangri01, 2025-08-17:
> hmmm... schaltest du opendtu aus, weil du am probieren bist? Ich habe die immer an... Und überhaupt, ziehst du den Stecker der opendtu? Hast du schon mal versucht nur 1 Solarflow anzuregeln? die [...]

Ja ich schalte die openDTU aus. Steht auch oben. Ich habe einen Wlan Plug Steckdose Installiert und diese schalte ich einfach im HA aus.

Muss mir mal Zeit nehmen und alle neu Starten und einen nach dem anderen SolarFlow Probieren

closed 2025-09-22 by github-actions[bot]