Zendure/Zendure-HA #654: Keine Entitäts-Werte nach Update

chipsn, 2025-08-12:
Nach dem Update von Version 1.0.46 auf 1.1.3 werden zwar alle meine Geräte (Hyper 2000, Akkus) angezeigt aber es kommen keine Entitäts-Werte an.
Bei Eingabe des Tokens kommt der Fehler: unique_id_mismatch
Und bei Aufruf der Konfiguration zeigt es den Fehler: Der Konfigurationsfluss konnte nicht geladen werden: 500 Internal Server Error Server got itself in trouble

CommanderROR9, 2025-08-12:
Integration einmal löschen und neu Aufsetzen sollte helfen

simonschlauri, 2025-08-12:
Erhalte einen Einrichtungsfehler. Eingegebene Werte sollten stimmen. Habe Integration gelöscht und neu aufgesetzt. Bringt nix. Habe es mit und ohne mqtt versucht, kein Unterschied.

Kannst du damit was anfangen?:

Logger: homeassistant.config_entries
Quelle: config_entries.py:749
Erstmals aufgetreten: 23:48:38 (3 Vorkommnisse)
Zuletzt protokolliert: 23:52:34

Error setting up entry Zendure for zendure_ha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 749, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/zendure_ha/__init__.py", line 21, in async_setup_entry
    await manager.loadDevices()
  File "/config/custom_components/zendure_ha/manager.py", line 134, in loadDevices
    self.api.Init(self.config_entry.data, mqtt)
    ~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/zendure_ha/api.py", line 85, in Init
    Api.mqttCloud.__init__(mqtt_enums.CallbackAPIVersion.VERSION2, mqtt["clientId"], False, "cloud")
                                                                   ~~~~^^^^^^^^^^^^
KeyError: 'clientId'

Logger: homeassistant.components.sensor
Quelle: helpers/entity_platform.py:843
Integration: Sensor ([Dokumentation](https://www.home-assistant.io/integrations/sensor), [Probleme](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+sensor%22))
Erstmals aufgetreten: 23:52:27 (2 Vorkommnisse)
Zuletzt protokolliert: 23:52:34

Platform zendure_ha does not generate unique IDs. ID Zendure Manager-available_kwh is already used by sensor.zendure_manager_available_kwh - ignoring sensor.Zendure Manager-available_kwh

Logger: homeassistant.components.number
Quelle: helpers/entity_platform.py:843
Integration: Zahl ([Dokumentation](https://www.home-assistant.io/integrations/number), [Probleme](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+number%22))
Erstmals aufgetreten: 12. August 2025 um 23:52:27 (2 Vorkommnisse)
Zuletzt protokolliert: 12. August 2025 um 23:52:34

Platform zendure_ha does not generate unique IDs. ID Zendure Manager-manual_power is already used by number.zendure_manager_manual_power - ignoring number.Zendure Manager-manual_power

Logger: homeassistant.components.select
Quelle: helpers/entity_platform.py:843
Integration: Auswahl ([Dokumentation](https://www.home-assistant.io/integrations/select), [Probleme](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+select%22))
Erstmals aufgetreten: 12. August 2025 um 23:52:27 (2 Vorkommnisse)
Zuletzt protokolliert: 12. August 2025 um 23:52:34

Platform zendure_ha does not generate unique IDs. ID Zendure Manager-Operation is already used by select.zendure_manager_operation - ignoring select.Zendure Manager-operation

Logger: homeassistant.config_entries
Quelle: config_entries.py:749
Erstmals aufgetreten: 12. August 2025 um 23:52:27 (12 Vorkommnisse)
Zuletzt protokolliert: 12. August 2025 um 23:52:34

Error setting up entry Zendure for button
Error setting up entry Zendure for number
Error setting up entry Zendure for select
Error setting up entry Zendure for sensor
Error setting up entry Zendure for switch
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 749, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/binary_sensor/__init__.py", line 139, in async_setup_entry
    return await hass.data[DATA_COMPONENT].async_setup_entry(entry)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 219, in async_setup_entry
    raise ValueError(
    ...<2 lines>...
    )
ValueError: Config entry Zendure (01K2G3SXREKJCMX0G2JG6EKGEC) for zendure_ha.binary_sensor has already been setup

CommanderROR9, 2025-08-12:
Ich bin da leider mein Experte. Hast Du den Api Key aus der App verwendet?

harrymayr, 2025-08-12:
Noch viel wichtiger. Hast du den API-Key des **Haupt-Accounst** benutzt? Der zweite Account wird nicht mehr benötigt!

chipsn, 2025-08-13:
Es läuft - Dankeschön 🙏