Zendure/Zendure-HA #659: [Bug] Instructions do not match what is asked

TheCarpetMerchant, 2025-08-13:
### Home Assistant Version

2025.7.3

### Zendure Integration Version

None

### Describe the bug

This is what I see after having clicked the MY button on the README :

[image]

This doesn't make any mention of an API key and doesn't look at all like what the instructions of the README talk about ?

### To Reproduce

NA

### Expected Behaviour

NA

### What device are you using?

SF2400 AC

### Log Extracts

```shell

```

TheCarpetMerchant, 2025-08-13:
As I said, the options you show do not exist. Clicking the options of the integration only shows this :

[image]

Checking the "mqttlog" checkbox and Submit does not change anything.

harrymayr, 2025-08-13:
> As I said, the options you show do not exist.

The reconfigure does. And what do you think how the local communication should happen? It uses the WiFi.

TheCarpetMerchant, 2025-08-14:
> > As I said, the options you show do not exist.
> 
> The reconfigure does (using the 3 dots). And what do you think how the local communication should happen? It uses the WiFi. So the integration nees this information.

This is incorrect, it shows the same thing as in my first screenshot (with info filled in already, obviously).

I have deleted and reinstalled the extension. I've noticed that it stops working after a while. Here we can see the battery output is frozen after a while. It does not correspond to the Zendure app data.[image]

FireSon, 2025-08-14:
@TheCarpetMerchant From the screenshot you supplied, it is clear that you have installed an older version of the Integration. This is the reason we ask you to fill out the version number. You can find that in "Devices & Services", and clicking on: "Zendure Home Assistant Integration". It should look like this:

[image]

You should upgrade to the latest stable release (1.1.3)! That will show the correct screens @harrymayr was talking about.

closed 2025-08-14 by TheCarpetMerchant