Zendure/Zendure-HA #673: Zendure manager stops ..

Spottttt, 2025-08-17:
Hello
Config: 2 hypers 2000.

When one hyper reaches the Soc limit  (high/low), then the Zendure manager stops and the other hyper does not start.

[image]

[image]

harrymayr, 2025-08-17:
Can you check, if the code change in manager.py which is described [here](https://github.com/Zendure/Zendure-HA/issues/646#issuecomment-3193899065) solve your Problem? This is around line 264.

Spottttt, 2025-08-17:
I modify :                     
                socLimit = d.socLimit.asNumber == 2  #(1 if isCharging else 2)

as you ask.
 Is it right ??

harrymayr, 2025-08-17:
Better this way (completely remove check on socLimit). Your modify may not work on minSoc (your issue). However, it's not yet clear to me, why this create that issue

[image]

Spottttt, 2025-08-17:
When i delete "socLimit = d.socLimit.asNumber == (1 if isCharging else 2)" then Zendure Manager does not start at all.

harrymayr, 2025-08-17:
You also need to change the line starting with
`if g.powerAvail ==` ...
Please check the code snippet above.

Spottttt, 2025-08-17:
All the changes are done : zendure manager start and is charging .

I wait soclimit (high/low) and  I will make a feed back.

Spottttt, 2025-08-17:
Héla !!!! Same behavior with change:    

[image]

Zendure manager stops ...

Zanooon97, 2025-08-17:
Hi, I have made some changes in the code, for cluster systems I have to put the socLimit again in the code otherwise it doesnt know witch device to take for the discharge. 

if (g := d.fusegroup) is not None and d.online:
                useDevice = abs(0.85 * maxPower) < abs(power) if d.powerAct == 0 else abs(0.80 * maxPower) < abs(power)
                totalKwh += d.availableKwh.asNumber
                
                # acMode: 1 = AC-Ladung, 2 = AC-Entladung
                acMode = getattr(d.acMode, "value", 0)
                isACChargingMode = acMode == 1
                
                _LOGGER.info(f"[DEBUG] Device {d.name}: acMode={acMode}, isACCharging={isACChargingMode}, isCharging={isCharging}, state={state}, socLimitRaw={getattr(d.socLimit, 'asNumber', None)}")

                socLimit = False
                if (isACChargingMode and isCharging and getattr(d.socLimit, "asNumber", 0) == 1):
                    socLimit = True
                elif (not isCharging and getattr(d.socLimit, "asNumber", 0) == 2):
                    socLimit = True
                
                #socLimit = d.socLimit.asNumber == 2 #(1 if isCharging else 2)
                if g.powerAvail == g.powerUsed or not useDevice or socLimit:
                    d.power_set(state, 0)
                else:
                    d.powerAvail = max(g.powerAvail - g.powerUsed, d.powerMin) if isCharging else min(g.powerAvail - g.powerUsed, d.powerMax)
                    g.powerUsed += d.powerAvail
                    maxPower += d.powerAvail
                    devs.append(d)

Spottttt, 2025-08-17:
@Zanooon97
Ok i will try it as soon as possible .
Thanks

Zanooon97, 2025-08-17:
whats the log?

FireSon, 2025-08-20:
@Zanooon97 Just try to catch up with this discussion. Can you explain what is wrong with the currenct code? If I look at the changes you have proposed to @Spottttt, I do not think those changes are ok:

```
        isMinSoc = getattr(d.electricLevel, "asNumber", 0) == getattr(d.minSoc, "asNumber", 0)
        isMaxSoc = getattr(d.electricLevel, "asNumber", 0) == getattr(d.socSet, "asNumber", 0)
```

The isMinSoc and isMaxSoc, is not always working, probably due to some internal rounding of the device. Which is exactly why I switched to socLimit, which indicates min and max soc in one entity.

Zanooon97, 2025-08-20:
I saw in the log of Spot that he has socLimit 0 with minSoc = maxSoc
probaby he is not getting the socLimit from his Hyper, i have searched in his logfile but there is no information about that

[image]

FireSon, 2025-08-20:
Ahh ok, great find! Although that is exactly why I started using the socLimit. It might help to also check the app, because it shows the 'correct' status...

Is he on the latest firmware versions? It seems a bit odd that his hypers do not have this entity...

Spottttt, 2025-08-20:
Hello @Zanooon97 @harrymayr 
I am not at home but from what I understood from yours interventions, it is that I would not have the right one.
Actually AC firmware 2.1.12
Master Firmware 2.1.24
Which one should I put and above all, is there a zero risk that Zendure does not control my installation after changes?

I will try this evening when I get back

FireSon, 2025-08-20:
@Spottttt basically you need the latest firmware for all Hyper2000 firmwares.

Master: v2.1.30
Bms: v1.0.17
AC: v2.1.14
Zenlink does not matter

closed 2025-08-22 by FireSon