Zendure/Zendure-HA #683: [Bug] v1.1.3 - 2400 AC - Battery drained while being idle

Youpimatin, 2025-08-19:
### Home Assistant Version

2025.7.3

### Zendure Integration Version

v1.1.3

### Describe the bug

First of all, thank you to the developers for this Home assistant integration.

I have a Solarflow 2400 AC, 1 battery, a Shelly Pro 3 EM.

When I was using the HEMS from Zendure, there was no battery draining when in Off state.

Now I use this integration.

When switching Zendure Manager mode from "Smart Matching" to "Off", there is always a 50W consumtion draining battery. (This is 1% SOC loss each 34 minutes)

acStatus stays at "1" instead of dropping to "0"

### To Reproduce

switch Zendure Manager mode from "Smart Matching" to "Off"

### Expected Behaviour

No Battery draining when in Off state

### What device are you using?

SF2400 AC

### Log Extracts

```shell

```

Youpimatin, 2025-08-20:
Thanks, I just tried to check :
- Offgrid plug is not used, and Smart Mode is not checked in the Zendure app
- Zendure Manager is changed to "Off"
- input limit is set to 0
- output limit is set to 0

AC mode set to "output mode" or "input mode", but AB3000 power stays around 50W... I don't understand, but if I can help by testing something else, or by sending logs, tell me.

FireSon, 2025-08-22:
Can you test this with version 1.1.4-pre2?

Youpimatin, 2025-08-22:
I'm pleased to announce that 1.1.4-pre2 has resolved this issue.
This is excellent, thank you very much.

FireSon, 2025-08-22:
Great 😄 I must say it was quite a puzzle to find...

closed 2025-08-22 by FireSon