Zendure/Zendure-HA #684: [Bug] switchCount resets when integration is updated

Youpimatin, 2025-08-19:
### Home Assistant Version

2025.7.3

### Zendure Integration Version

v1.1.3

### Describe the bug

switchCount resets when integration is updated
(I think it was also the case with "Total Charged" and "Total Discharged", but not anymore ?)

### To Reproduce

- update integration

### Expected Behaviour

Do not reset the value

### What device are you using?

SF2400 AC

### Log Extracts

```shell

```

harrymayr, 2025-08-19:
What is your interest in this value for long term statistics?

FireSon, 2025-08-20:
Thanks for the feedback, we will fix it in the next release!

closed 2025-08-20 by FireSon

harrymayr, 2025-08-20:
> Do you think it's too hard to achieve or not recommended ?

Not a problem, just want to understand the usecase. @FireSon made already changes, which will be available in the next (pre) release.