Zendure/Zendure-HA #686: [Bug] Zendure doesn't keep Apartment at 0W after reaching 100% SoC

Tronnic, 2025-08-20:
### Home Assistant Version

2025.7.2 Core / 2025.08.1 Supvervisor

### Zendure Integration Version

v1.1.3

### Describe the bug

Hello everyone, I installed my Zendure SolarFlow800 Pro yesterday evening. So today is it's first day of running.
I use a Shelly 3EM Pro to measure my home power usage. The Zendure Addon is configured to use it. 

[image]

If you look at the graph at the bottom, while the battery was charging, my home power usage was kept pertty much exactly at 0W which I was VERY happy about. 

However, after reaching 100% SoC, it's really having trouble keeping the grid power usage at 0W. It says it's only pruducing 70W Solar, which is clearly wrong. The sun is shining bright and the modules should deliver around 1200W at this time. 

Is there something I did wrong? I'm somewhat dissapointed that I use 100-200W even if the sun is shining bright. 

SolarFlow 800 Pro:
[image] [image]
[image] [image]

Battery:
[image]

Zendure Manager:
[image]

Thanks for any help I can get. The Firmware of the Zendure is Up-to-date. 

### To Reproduce

1. Install HACS Zendure-HA
2. Setup SolarFlow800 Pro
3. Set everythting up
4. Let device run normally

So basically nothing special

### Expected Behaviour

After the Battery SoC Reaches 100%, the power of solar should still keep the house at 0W, or even pass trough full solar power. Whatever works. 

### What device are you using?

SF800 Pro

### Log Extracts

```shell
All I could find when searching for "Zendure" in the HA Logs:

Logger: custom_components.zendure_ha.device
Source: custom_components/zendure_ha/device.py:490
integration: Zendure Home Assistant Integration (documentation, issues)
First occurred: 1:13:33 PM (14 occurrences)
Last logged: 1:17:35 PM

HttpPost error SolarFlow 800 Pro Cannot connect to host 192.168.1.147:80 ssl:default [Connect call failed ('192.168.1.147', 80)]!

----

Logger: custom_components.zendure_ha.device
Source: custom_components/zendure_ha/device.py:479
integration: Zendure Home Assistant Integration (documentation, issues)
First occurred: 1:13:22 PM (48 occurrences)
Last logged: 1:17:35 PM

HttpGet error SolarFlow 800 Pro Cannot connect to host 192.168.1.147:80 ssl:default [Connect call failed ('192.168.1.147', 80)]!

----

Logger: py.warnings
Source: custom_components/zendure_ha/entity.py:13
integration: Zendure Home Assistant Integration (documentation, issues)
First occurred: August 19, 2025 at 7:33:01 PM (1 occurrence)
Last logged: August 19, 2025 at 7:33:01 PM

/usr/local/lib/python3.13/site-packages/stringcase.py:247: SyntaxWarning: invalid escape sequence '\W' return re.sub("\W+", "", string)
```

Tronnic, 2025-08-20:
Okay, so after looking at my own screenshots, I noticed that "Ausgangslimit" and "Eingangslimit" is set to very low. I set this to the maximum yesterday. Also I monitored the whole day today, theses limits were not that low while charging. See the graph here (my idle grid usage of my home is at about 250-300W)

[image]

It seems to be working now after setting these sliders to max agian

[image]

I will report back if anything changes.

Tronnic, 2025-08-20:
Allright, so after a few minutes the slider went back to 33W. Why? How can I keep it at max? When set to max, everything works. I mean sure I could create an automation for that, but I guess thats not pracitcal or good for flash.

[image]

Just while typing this, the slider once again changed. I guess it represents what the battery should be set at to reach 0, which is clearly not enough. 

Is there a way to fix this?

Zanooon97, 2025-08-21:
okei I dont knew that, that non ac load capable devices exist here by zendure.
So we need to change it:

    async def power_get(self) -> int:
        """Get the current power."""
        if not self.online or self.outputHomePower.state is None or self.gridInputPower.state is None:
            return 0
        self.powerAct = self.packInputPower.asInt - self.outputPackPower.asInt
        if self.powerAct != 0:
            self.powerAct += self.solarInputPower.asInt
        if self.powerAct == 0:
            self.powerAct = self.outputHomePower.asInt
        return self.powerAct

In my opinion it is better to calculate with outputHomePower instead of solarPower, the solarpower can be 280W and the effective output power is 250W

Zanooon97, 2025-08-21:
@FireSon Hello, whats the reason about this exclusion of solarpower when battery is inactive?

Tronnic, 2025-08-22:
My solution for now is to set bypass to always on.

[image]

When battery is full, all the power will go into the grid. It produces more heat, but at least I don't loose energy. I would love to have a setting at some point to set passtrough whenn full to a limit of 100W or something. This way, small fluctuations don't cost anything.

Zanooon97, 2025-08-22:
@Tronnic https://github.com/Zendure/Zendure-HA/issues/646

Zanooon97, 2025-08-25:
look at the pullrequest from harry, and update to the latest pre 2 version.
https://github.com/Zendure/Zendure-HA/pull/702

Zanooon97, 2025-09-01:
@spirann make a debugg log when you got this problems

closed 2025-10-07 by github-actions[bot]

spirann, 2025-10-15:
I'm running 1.1.4-pre7 and don't have the bug anymore.