Zendure/Zendure-HA #696: v1.1.4-pre1 | Houston we have a serious problem! | HA-update 'core 2025.8.3' makes Z-HA unusable! | DO NOT UPDATE !

Linus86529, 2025-08-22:
@FireSon 
After updating to Home Assistant core 2025.8.3, version v1.1.4-pre1 has become completely unusable. Even restarting HA or the system does not help. Z-HA v1.1.4-pre1 remains unusable.

I will try to see if a complete reinstallation of v1.1.4-pre1 can fix the problem. If that doesn't help, I will revert to HA Core 2025.8.2.

If you don't want to take any risks, it is better not to install the HA update Core 2025.8.3.

Firson and Harrymayr, please find out what is causing this.

 - 1st hint to the problem
[image]

 - 2nd ERROR overview
[image]

 - In the attached text file you'll find the content of the 3 above listed ERROR titles
[Manuskript___HA-Update_2025.8.3_kills_the_Z-HA-Integration___It_makes_it_totaly_ununsable.txt](https://github.com/user-attachments/files/21934236/Manuskript___HA-Update_2025.8.3_kills_the_Z-HA-Integration___It_makes_it_totaly_ununsable.txt)

 - 3rd Z-HA Integration device list - show not loaded
[image]

 - 4th All entities on all devices show ''
[image]

Linus86529, 2025-08-22:
Hello @MacdiverOne, you saved my day! I have no explanation for the error with HA Core 2025.8.3 and my v1.1.4-pre1 installation.

In any case, your tip about Pre2 was worth its weight in gold.
Now Pre2 is running on the latest HA Core as if nothing had happened. Thank you very much!

MacdiverOne, 2025-08-22:
I would keep it open and let @FireSon decide.

phlexss, 2025-08-22:
I was running 1.1.4-pre2 but at some point today had to re-pair my Hyper2000 because nothing was happening anymore in HA......after re-pairing and removing/adding my Hyper2000 in HA everything worked again.....

DerUnwissende, 2025-08-22:
Can't agree. My 114pre-1 is running well and all my sensors are available

FireSon, 2025-08-22:
Just upgraded to 2025.8.3 without any problems. From the logfile I can see that a file might have been corrupted, but difficult to tell. Close this for now. Please add any feedback to the release to the discussion #695

closed 2025-08-22 by FireSon