Zendure/Zendure-HA #705: Zendure SolarFlow 1200 Integration Home Assistant - Bluetooth ???

mangri01, 2025-08-25:
### Describe the feature

Hallo, ich habe schon länger versucht mit dieser Zendure Integration meine 4 Stück Solar Flow zu steuern, Erfolglos.

Nun habe ich, da ich keine Anleitung gefunden habe etwas gefunden was was ich nicht gewusst habe.

Anscheined muss man die SolarFlow per Bluetooth si einstellen, dass Sie ihre Daten an das MQTT senden.

**

[image]
[image]
[image]

[image]
[image]

**

Mein System
HA 2025.8.3
MQTT ist Installiert und Funktioniet
MQTT User und PW vorhanden
Zendure Hacs Version 1.1.4 pre 2 ist Installiert und alle Werte werden der Geräte Solar Flow 1200 und der 4 Akks (2 x AB 2000 und 2 x AB 1000) werden Angezeigt.
Zendure Token ist eingetragen
Sensor sensor.power.actual habe ich per Helfer erstellt und er zeigt negative und positive Werte an. 
Häckchen bei Lokal Mosquitto MQTT Addon verwenden ist gesetzt und alles Konfiguriert

4 x Solar Flow 1200, je SolarFlow 1 AB Akku
2 x HMS 1800 Wechselrichter
1 x Shelly 3em pro per LAN Verbunden

Ich habe die Verbindungsart in der HA Zendure auf Lokal gestellt, aber es wird an den Hubs nicht die Verbindungsart zu Lokal gewechselt (siehe Screen).

### What device is this for?

Hub1200

mangri01, 2025-08-25:
> Ich habe immer wieder gelesen, das bei "mqttserver" die IP-Adresse deines Mosquitto-Brokers rein muss, nicht dieser Hostname, der bei dir steht. Ich habe die Integration mit einem Hub2000 auch so am Laufen und mit Version 1.1.4-pre2 funktioniert dies auch. Ich musste nach dem Umstellen auf Lokal den Hub noch händisch neu starten mittels 6 Sekunden den Button drücken. Bluetooth ist tatsächlich nur zum setzen der lokalen Verbindung erforderlich, die Daten werden anschließend über das angegebene WLAN an den lokalen MQTT-Broker übertragen. Du solltest dann am MQTT-Explorer auch mehrere neue Einträge sehen, die nicht unter "iot" stehen. Da sollten auch die tatsächlichen Werte auftauchen.

Danke für die Info.Ich habe die P Adressse des Mosqitto Brokers eingetragen.
Wie kann man das mit dem Bluetooth machen, das setzen der lokalen Verbindung??? Man muss glaube ich dem SoalrFlow die Anweisung geben die Daten an das MQTT zu senden. Nur wie geht das...

Ich möchte weg von der Cloud..

dozi-84, 2025-08-25:
Das passiert, wenn du bei Verbindungsart "Lokal" auswählst. Bei der neuen Version sollte auch eine Meldung im HA auftauchen, ob dies erfolgreich war oder nicht. Anschließend müsste der Hub1200 auch neu gestartet werden nehme ich an. Danach sollten mehr Daten im Broker zu sehen sein, wenn sonst alles passt natürlich.

mangri01, 2025-08-25:
> Deine 4 SF 1200 müssen in BT Reichweite deiner HA Installation sein. Dann musst du nur jeden auf Lokal stellen, das funktioniert normalerweise. Ich habe aber z.B. das Problem, dass die BT Verbindung immer wieder abbricht. Es gibt auf jeden Fall mit der 1.1.4-pre2 eine Benachrichtigung, um die Umschaltung geklappt hat (oder auch nicht). Falls es nicht funktioniert, könnte es bei der pre2 noch nötig sein, wieder auf cloud zu wechseln und dann wieder auf lokal. In der in den Startlöchern stehenden pre3 muss man nur auf den Verbindungs Reset Knopf gehen, dann wir die Umstellung der Verbindung nochmals versucht. Musst du im Zweifel so oft machen, bis es geklappt hat.

Danke, ich verstehe das mit dem Blutooth nicht. Wie verbindet sich das Bluetooth mit den SolarFLow.

harrymayr, 2025-08-25:
Meine HA Installation ist auf eine Raspberry Pi, der hat eine BT Schnittstelle. Die verbindet sich dann automatisch mit dem Hyper (sofern in Reichweite)

mangri01, 2025-08-25:
> Es muss natürlich in HA auch [bluetooth](https://www.home-assistant.io/integrations/bluetooth/) installiert sein

Hab ich 

[image]

harrymayr, 2025-08-25:
Wenn die Hubs in Reichweite sind, solltest du sie im Advertisement Moniter von Bluetooth sehen, als Zen... Geräte. Im Visuellen Monitor müssen sie grün sein, damit das mit dem local mqtt klappt.

mangri01, 2025-08-25:
Super, ich danke dir. 
Ich werde es testen, sobald ich wieder aus dem Urlaub zu Hause bin.
Mitte nächste Woche.

Sepp2812, 2025-08-25:
> Deine 4 SF 1200 müssen in BT Reichweite deiner HA Installation sein. Dann musst du nur jeden auf Lokal stellen, das funktioniert normalerweise. Ich habe aber z.B. das Problem, dass die BT Verbindung [...]

und das ist das gefährliche daran, wenn die BT Verbindung abbricht und sich nicht mehr neu Verbindet, dann regelt  z.B der Hyper nicht mehr

harrymayr, 2025-08-25:
> und das ist das gefährliche daran, wenn die BT Verbindung abbricht und sich nicht mehr neu Verbindet, dann regelt z.B der Hyper nicht mehr

Das ist Quatsch, die BT Verbindung wird nur gebraucht, um den MQTT Server im Hyper umzustellen, die Regelung erfolgt weiterhin über WLAN.

mangri01, 2025-08-25:
Ich habe gerade bei HA Bluetooth, Konfigurieren, Advertise Monitor bei Visualisierung gesehen, dass meine 4 Stück SolarFlow 1200 angezeigt werden. 

Da ich in der Garage ein Bluetooth Gerät habe, das für das Garagentor verantwortlich ist.

Also kennt der HA die SolarFlow ja. Wie muss ich jetzt vorgehen, dass er von Cloud auf Lokal wechselt.

[image]

Sepp2812, 2025-08-25:
> > und das ist das gefährliche daran, wenn die BT Verbindung abbricht und sich nicht mehr neu Verbindet, dann regelt z.B der Hyper nicht mehr
> 
> Das ist Quatsch, die BT Verbindung wird nur gebraucht, um den MQTT Server im Hyper umzustellen und wird nur da einmalig gebraucht, die Regelung erfolgt weiterhin über WLAN.

muss doch nochmal nachfragen, das heist wenn eine lokale Verbindung besteht, könnte ich den Hyper vom Internet sperren ?

[image]

mangri01, 2025-09-03:
Ich konnte meine 4 Stück Solar Flow 1200 mit einem UGREEN Bluetooth Adapter 5.3 auf die Lokale Verbinung erfolgreich umstellen.

[image]

Im MQTT Exlorer sind die 4 Stück Solar Flow ganz unten zu sehen. 

[image]

Nun noch ein paar Fragen für das weitere vorehen der Einstellungen der Solar Flow

1. Da ich bisher eine openDTU on Battery nutze habe ich die Ausgangsleistung bei den Solar Flow auf 900 gestellt.
2. Ich habe 4 Solar Flow und vier Akku. Die Sicherungsgruppe habe ich bei dem ersten Solar FLow Sicherungsgruppe max. 3600 eingestellt. Alle weitere Solar Flow auf Part of Solar Flow 1...

[image]

[image]

Nun meine Frage, muss man das Ausgangslimit der Solar Flow auf null setzen.
Oder wie habt ihr eure Solar Flow eingestellt.

DANKE

mangri01, 2025-09-03:
> Sicherungsgruppe ist OK, wenn das deine Installation so erlaubt. Bzgl. Ausgangslimit: Im Prinzip ist dieser Wert für die Manuelle Entladung gedacht, und wird (eigentlich) bei der Smarten Leistungsregelung nicht berücksichtigt. Ich habe aber schon bei diversen Anwendern erlebt, dass es bei der HUB1200/2000 Familie sinnvoll ist, auf 0 zu stellen.

Danke 
Irgendwo scheint liegt noch ein Fehler. Es werden nur 2 von den vier Solar Flow geregelt. Muss die alle morgen mal neu Starten.

mangri01, 2025-09-03:
> Bei der 1.1.4-pre2 werden auch nicht immer alle verwendet. Schau mal in die [Wiki ](https://github.com/Zendure/Zendure-HA/wiki/Power-distribution-strategy)

Ok, danke. Welche Version Zendure soll ich in HA Installieren damit alle vier Solar Flow geregelt werden??