Zendure/Zendure-HA #709: [Bug] Integration needs reload before changing action

Martinvdm, 2025-08-26:
### Home Assistant Version

2025.8.2

### Zendure Integration Version

1.1.4-pre2

### Describe the bug

After upgrading to 1.1.4-pre2 it seems that de integration needs reload before there is any activity. for instance, the status is inactive, but there is >400W PV Solar to grid measured and also viewable in the sensors. Zendure is not changing to Charging dispite the setting Smart Matching. After reloading the integration, is suddenly it starts charging.

### To Reproduce

overnight not changing status

### Expected Behaviour

automatically change to charging

### What device are you using?

SF2400 AC

### Log Extracts

```shell
Will try to gather it
```

Sepp2812, 2025-09-14:
Das ist bei mir auch so, muss manuell auf neu laden gehen, dann funktioniert das Akkuladen bei Überschuss wieder, Entladen funktioniert immer ( 2× Hyper 2000)

Martinvdm, 2025-09-14:
@Sepp2812  please keep it in English. Not to difficult isn’t ….