Zendure/Zendure-HA #710: [Feature] Control or override Smartmode / manual charging in Smartmode

dagsta, 2025-08-26:
### Describe the feature

I have the following requirement:

I am generally very satisfied with the Smart Mode on the Hyper 2000, as it responds very quickly to changes and the control system feels local and therefore very fast.

However, I would still like to be able to override Smart Mode and force the Hyper 2000 to charge AC. It would therefore be great if I had the option in HA to either change or override Smart Mode in order to control manual charging (regardless of current household consumption).

Is this already possible in some way, or would it be possible to implement it?

Translated with [DeepL.com](https://www.deepl.com/?utm_campaign=product&utm_source=web_translator&utm_medium=web&utm_content=copy_free_translation) (free version)

### What device is this for?

Hyper2000

harrymayr, 2025-08-26:
You should not use any energy modes or HEMS if you want to control the Hyper via the integration. And I assumed you use the Smart matching mode in the integration.

dagsta, 2025-08-27:
Exactly as I described, I find the control via Hyper great in normal cases because of the low latency. That's why I'm asking whether I can somehow switch Smart CT Mode on and off via HA.

Andreasq22, 2025-08-28:
> > Is this already possible in some way, or would it be possible to implement it?
> 
> Sure, you can set the Zendure Manager to Manual mode and use the Manual power with a negative value to charge the Hyper.

this would be great, but it ist not functioning

harrymayr, 2025-08-28:
> but it ist not functioning

Of course it is! To check why it doesn't work on **your** side, I need a debug log to check.

harrymayr, 2025-08-30:
Debug-Logging could be enabled with the 3 dots. After that, restart HA, after restart is finished, do the change to the manual mode. After a few minutes you can deactivate the debug logging. Your browser will download a file. This file you can provide here.

But first? Did you select a fuse-group for your Hypers? This is necessary to use any of the functions where the Zendure Manager is not set to OFF.

[image]

Andreasq22, 2025-08-30:
> Debug-Logging could be enabled with the 3 dots. After that, restart HA, after restart is finished, do the change to the manual mode. After a few minutes you can deactivate the debug logging. Your [...]

The three hypers are on its own phase. This is what i configured in the Integration. Everything works fine, but Not the manual mode... current i set the Mode to off and the three hypers to 1200 Watt Input mode. This works and i think, This is what the Integration have to do, when i set the mode to manual and -3600 watt

Mart-Muc, 2025-09-06:
Im afraid, you cannot switch on/off HEMS with the the integration!
which is necessary for local control with the Zendure Manager Control

this would be a gooooood [feature] for the next release