Zendure/Zendure-HA #711: [Bug] Low solar input causes battery cycling at low SoC instead of passthrough

Tronnic, 2025-08-27:
### Home Assistant Version

Core 2025.8.3 / Supervisor 2025.08.1

### Zendure Integration Version

Master + Changes from PR702

### Describe the bug

I've noticed an issue where the battery enters a constant charge/discharge loop when solar generation is lower than the household consumption.

With minimum SoC set to 10%, the battery constantly charges to ~12% and then discharges back to 10%. This loop continues as long as there is a power deficit. The system fails to switch to a stable passthrough/bypass mode.

As you can see in the attached image, this behavior puts many unnecessary cycles on the battery.

[image]

### To Reproduce

Normally running the system

### Expected Behaviour

The expected behaviour is that the battery should not attempt to charge at all. It should switch to passtrough-mode and only start charging when the solar input surpasses the current grind-need.

### What device are you using?

SF800 Pro

### Log Extracts

```shell

```

Zanooon97, 2025-08-27:
Da gibt es keine einstellung die ein bypass aktiviert und nur überschüssige energie in den Akku lädt, HEMS macht es da schon besser in diesem Fall.
Da müsste mann bei Zendure fragen ob sie so ein setting bereitstellen. @harrymayr  liege ich da falsch? ich habe auch schon danach gesucht.
Hems sagt, 
1. Haushalt
2. Akku
3. Netz

agusie, 2025-08-27:
I have the same behaviour here. 2 Hyper2000 with one battery each and 2x500W solar panels on each Hyper. I have tried different settings in the integration and even if I activate HEMS in the Zendure app, I still get the same... When "SoC min" is reached all the solar energy is fed to the batteries until they gain +2% and then the batteries discharge again... it keeps doing that all day long lately as it has been quite cloudy.

MrSuika, 2025-08-30:
I also use the HEMS for my SF 800 Pro to have better statistics for each panel and quicker online acess to them than in HA, and at first the automatic bypass worked without any issues, but now that constantly cycles between bypass and battery charging, like @agusie discribed, when the battery is at SoCmin, I think it didn't do that when I got it a month ago, might have something to do with batterie calibration tho, because the online thing i did, was removing the 800 pro from my 2 AB 2000 batteries to connect them to my HUB 2000 and microinverter once more to delete the data on them, then I put the 800 Pro back on the batteries and called it a day.

But I have also contacted Zendure support of this issue Already, because that can't be healthy for the bypass relais and I don't want to RMA a device which is barely older than a few month.

Another weird thing I noticed tho, it isn't always cycling through the modes, sometimes it's fine as you can see in the Screenshots, which is weird in itself. 
I will try to fully charge rhe batterie and see if that helps. 

[image]
[image]
[image]

MrSuika, 2025-08-30:
> I also use the HEMS for my SF 800 Pro to have better statistics for each panel and quicker online acess to them than in HA, and at first the automatic bypass worked without any issues, but now that constantly cycles between bypass and battery charging, like [@agusie](https://github.com/agusie) discribed, when the battery is at SoCmin, I think it didn't do that when I got it a month ago, might have something to do with batterie calibration tho, because the online thing i did, was removing the 800 pro from my 2 AB 2000 batteries to connect them to my HUB 2000 and microinverter once more to delete the data on them, then I put the 800 Pro back on the batteries and called it a day.
> 
> But I have also contacted Zendure support of this issue Already, because that can't be healthy for the bypass relais and I don't want to RMA a device which is barely older than a few month.
> 
> Another weird thing I noticed tho, it isn't always cycling through the modes, sometimes it's fine as you can see in the Screenshots, which is weird in itself. I will try to fully charge rhe batterie and see if that helps.
> 
> <img alt="Image" width="1280" height="2000" src="https://private-user-images.githubusercontent.com/214804291/483854219-acdcd2aa-743a-4f40-9385-dfe642ba298a.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTY1MzYxNDAsIm5iZiI6MTc1NjUzNTg0MCwicGF0aCI6Ii8yMTQ4MDQyOTEvNDgzODU0MjE5LWFjZGNkMmFhLTc0M2EtNGY0MC05Mzg1LWRmZTY0MmJhMjk4YS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwODMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDgzMFQwNjM3MjBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hODYzNjc1N2ZkMzE2MGYwYTNiMDhhMjZmNGI5MDg2ZmE0ZTEyMjFiNGQ4NTg3NDRmN2JkOTk5YmIwY2FkMmI1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.XAIhgRTEQ_TkzzUFZgCVjvP04vkaDyr1oDPGqFAniTc"> <img alt="Image" width="1280" height="2000" src="https://private-user-images.githubusercontent.com/214804291/483854220-2f2e34db-82de-4261-b3c6-a7efda1f8719.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTY1MzYxNDAsIm5iZiI6MTc1NjUzNTg0MCwicGF0aCI6Ii8yMTQ4MDQyOTEvNDgzODU0MjIwLTJmMmUzNGRiLTgyZGUtNDI2MS1iM2M2LWE3ZWZkYTFmODcxOS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwODMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDgzMFQwNjM3MjBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04YWJkYTJhMmEyNDNlNzExZDhkNzZjYTYzNzJiNmZlM2RkOTYyMDZkYjdlNWYwNTk4MmNlNWYxM2ZiZWRhZjc4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.oKFETEv-ScbkcvBVD6SAXuMPrPSNtJk4wzJaRe_Aifg"> <img alt="Image" width="1280" height="2000" src="https://private-user-images.githubusercontent.com/214804291/483854221-28c679e4-e8d1-420b-8926-a74f686bb5ab.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTY1MzYxNDAsIm5iZiI6MTc1NjUzNTg0MCwicGF0aCI6Ii8yMTQ4MDQyOTEvNDgzODU0MjIxLTI4YzY3OWU0LWU4ZDEtNDIwYi04OTI2LWE3NGY2ODZiYjVhYi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwODMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDgzMFQwNjM3MjBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wZTRhZGZiODg4YzQ1OTFhNjRhOGJlNTk5ZTcxNTNkNTdiMzkxNWE4NTJhNzNiOTY0MDY3Y2FmN2I5MTlmZThhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.3j5jME7i7IfO-8_uI-DmEM7IVaz2wJuWPNsYHhn2iCk">

Ok never mind, i've found out what the issue here was, I though when the device is connected to the HEMS it doesn't take any commands from HA and for the device itself that seems to be the case, but my Zendure Manager was set to Smartmatching and apparently that can sent commands to the Device, even if it's included into the HEMS @harrymayr @FireSon maybe you can take that into consideration for a future update and check for HEMS status, if possible, before the Zendure Manager uses a device to make it more idiot proof 😅.

cybe, 2025-09-02:
> I've noticed an issue where the battery enters a constant charge/discharge loop when solar generation is lower than the household consumption.

I'm also affected. Happy to help out by testing some changes.

Zanooon97, 2025-09-13:
i made a pull request for the newest version yesterday, i changed also a littlebit the calculation of power because for my devices it wasnt working. 
https://github.com/Zendure/Zendure-HA/pull/771

cybe, 2025-09-13:
Thank you @Zanooon97 :)

>  is the new pre 4 working for you ??

Unfortunately not, the behavior was very weird: The `output_home_power` was somehow capped at 50 Watts, even though the home usage was like 300 Watts. Also the `solar_input_power` and the battery charging via `output_pack_power` fluctuated in quite a lot. I rolled back to an older version.

Didn't had the time to write a proper bug report yet. Maybe tomorrow I'm going to test the current master branch. There were some fixes applied since pre4.

harrymayr, 2025-09-13:
> There were some fixes applied since pre4.

Add also PR #775, otherwise there is a weird behavior if you have large solarInputPower.

cybe, 2025-09-13:
Will do :)

Zanooon97, 2025-09-14:
> Thank you [@Zanooon97](https://github.com/Zanooon97) :)
>
>
> Unfortunately not, the behavior was very weird: The `output_home_power` was somehow capped at 50 Watts, even though the home usage was like [...]

this capping at 50 watts should be fixed it was a never ending starting device hehe

cybe, 2025-09-14:
> Will do :)

fyi, continued this topic in [#775 ](https://github.com/Zendure/Zendure-HA/pull/775#issuecomment-3289459992)

Tronnic, 2025-10-15:
Yes it is still present as far as I know. And very annoying since winter is coming. I'm losing so much cycles and energy right now. Starting to loose hope and unsure if it was a good idea to go for zendure. :/  I even donated 20€ to you guys. A little fix for this would be great.