Zendure/Zendure-HA #717: v1.1.4-pre2 | 3rd: change of SOCmin or SOCset does not update "Verfügbare Energie"

Linus86529, 2025-08-27:
- Is it possible that devices in standby mode do not recalculate the “available energy” value when, for example, SOCmin is changed?
 - For devices that are in the process of discharging, is “available energy” always recalculated as soon as the SOCmin changes?

 - Doesn't that inevitably lead to incorrect sequence control of multiple devices?

initial situation:

Hyper2000: 'Entladevorgang'
  Akku-Pack (2 x AB2000)
  'Verfügbare Energie' = 1,613 kWh
  batSOC = 52%
  SOCmin = 10%
  SOCset = 90%

Hub 1200: Standby
  Akku-Pack (4 x AB1000)
  'Verfügbare Energie' = 1,267 kWh
  batSOC = 53%
  SOCmin = 20%
  SOCset = 80%

 - Change to Hub1200 (in standby mode):
New target value should be SOCmin = 10%.

The ‘available energy’ = 1.267 kWh has not changed.
I expected (53% - 10%) * 38.4 Wh/% = 1652.2 Wh.

I waited up to 30 minutes for the recalculation, but nothing happened.
Then I tried setting the values for SOCmin to 30%, then 40%, then 50%, and finally 70%. The “Available Energy” value never changed.

I then set SOCmin to 10% for the next day. The “Available energy” value did not change this time either.

Then I wanted to know what impact the changes would have on the Hyper2000.

 - Change to Hyper2000 (during discharge process):
The new target value for SOCmin is 20%, previously it was 10%.

The recalculation of “Available Energy” for the Hyper2000 takes a few seconds. Since the current value of the Hyper2000 is lower than that of the Hub1200,
the Hyper goes into standby mode and the Hub1200 starts the discharge process. When switching, the “Available Energy” value for the Hub1200 is immediately recalculated so that the previous setting of SOCmin = 10% now displays the correct “Available Energy”.

After changing what I previously did on the Hub1200 on the Hyper2000 with the SOCmin, the “Available Energy” value for the Hyper2000 is not recalculated. Could it be that no recalculations of the “Available Energy” value are performed when a device is in standby mode?

Who else has ever seen anything like this?

harrymayr, 2025-08-27:
SocSet is not part of the calculation, but you are right, updateing minSoc changes not immediatly Available Energy -> PR #718

FireSon, 2025-09-09:
Should be fixed in the next version 1.1.4-pre3

closed 2025-09-09 by FireSon