Zendure/Zendure-HA #726: [Bug] 1.1.4 pre2 won't start in "slimme matching"

pfwolbers, 2025-08-30:
### Home Assistant Version

2025.8.3

### Zendure Integration Version

V1.1.4 pre2

### Describe the bug

When the battery is empty (10%) in the night it won't start changing in the morning in " slimme matching" modus.
I've to put it in manual modewith a negative manual value to start charging. Then put it back in  manual mode and set the manual value to 0.0.
Then it will start again with the Charing in " slimme charging".

### To Reproduce

See description 

### Expected Behaviour

It should start charging when there is solar power available in the morning

### What device are you using?

Hyper2000

### Log Extracts

```shell

```

pfwolbers, 2025-08-30:
Today I had the same issue. It is a cloudy day, and the "slimme matching" stopt working (battery not empty). I had to restart the integration to get it working again.

harrymayr, 2025-08-31:
Do you have the solar panels direct connected to the Hyper, or do you use AC charge / discharge with a PV inverter direct connect to your house grid?

pfwolbers, 2025-08-31:
Not direct connected. I have a seperate inverter.

Tatayer58, 2025-08-31:
Bonjour, j'ai le même problème chez moi. Je n'avais pas le souci avec la prev 1 si cela peut aider.

Mediacj, 2025-08-31:
I have the exact same problem with the behaviour in “smart charging” when using directly connected solar panels.
I have to switch to another mode, for instance “smart charging”, then it starts, and afterwards I switch back to “smart matching”.

Zanooon97, 2025-08-31:
In the smart matching mode there are missing the startsequence for the negative p1 value, charging. 

[image]

pfwolbers, 2025-09-04:
Ok, nice. When will there be a release so i can download it easily?

FireSon, 2025-09-09:
This problem is fixed in version 1.1.4-pre3 which will be available soon!