Zendure/Zendure-HA #731: [Feature] 2400 AC - Off-grid plug settings

Youpimatin, 2025-08-31:
### Describe the feature

Is there a way to have access to these settings :
[image]

[image]

### What device is this for?

SF2400 AC

Youpimatin, 2025-09-09:
Wait, I already (1.1.4-pre2) have a "gridOffMode" entity, on my system, it's value can be :
0 : Off-grig plug is ON (Normal mode)
1 : Off-grig plug is ON (Eco mode)
2 : Off-grid plug is OFF (Zendure app : Smart mode OFF)

It reflects the off-grid plug state, but there is no entity to actually CHANGE these modes, do you have one ?

The only way I see to change the mode is to use the Zendure app.

FireSon, 2025-09-09:
In version 1.1.4-pre3 this will be possible

closed 2025-09-09 by FireSon