Zendure/Zendure-HA #746: [Feature] HEMS control switch

Mart-Muc, 2025-09-06:
### Describe the feature

Könnt ihr in der nächsten release bitte einen switch hinzufügen, um ein Device der HEMS Kontrolle zuzufügen und auch wieder auszuschalten.

(Um evtl. den Zendure Manager zu disablen, wenn HEMS aktiv, oder eine Warnung auszugeben)

Aber nun zu meinem  Fall:
Ich habe 2 SF Geräte, eines mit AB2000, das andere mit vorgeschalteter DC Batterie.

Ich möchte kontrollieren bzw priorisieren können, welche Batterie entladen wird- die automatische Aufteilung von HEMS funktioniert nicht, da der Batteriestand der vorgeschalteten 'nichtzendure' Batterie nicht einfließt.

Ebensowenig kommt der Zendure Manager der Integration mit diesem Setup zurecht.

### What device is this for?

SF800 Pro

### Other Information

a) SF800pro mit AB2000
b) SF800 mit Storcube Batterie am Solareingang

Mart-Muc, 2025-09-10:
What a pity... so im using the less elegant way with a shelly/tasmota plug :-(

...anyway...
What about Passthrough- is there a toggleswitch possible?
This week the weather is grey in munich and i want to force my SF800pro to pass all solarpower

Mart-Muc, 2025-09-14:
The use of a smartplug is not working!
because of available solarpower (by the dc battery) the device is always online and taking into account by calculating the power distributen.

Therefore 
I NEED a HEMS SWITCH!!!

harrymayr, 2025-09-14:
> wie funktioniert der ZendureApp API call ??
> 
> mit curl ist das ja für mich dann einfach zu realisieren.

Dazu müsste es in der API implementiert sein. Und so wie ich das verstehe, ist das nicht der Fall (und nicht geplant von Zendure?).

harrymayr, 2025-09-14:
> > wie funktioniert der ZendureApp API call ??
> > mit curl ist das ja für mich dann einfach zu realisieren.
> 
> Dazu müsste es in der API implementiert sein. Und so wie ich das verstehe, ist das nicht der Fall (und nicht geplant von Zendure?). Ich habe aber keine Ahnung, wie das in der App implementiert ist. Da müsste jemand mal ran, der sich mit Reverse Engineering von Apps auskennt.

Möglicherweise hilft da WireShark und die App übers WLAN weiter. Wird aber wahrscheinlich über https gehen und ist dann verschlüsselt.