Zendure/Zendure-HA #751: 📧 Version française

Yann33l, 2025-09-09:
📧 Version française

Bonjour,

Peut-être que ma question repose sur un simple a priori, mais je préfère la poser tout de même.

Dans le cas de deux SF2400 AC :
Lors de la charge (et probablement aussi lors de la décharge), si l’intensité dépasse la limite maximale fixée pour une batterie, la seconde batterie prend le relais. J’ai alors l’impression que la puissance est immédiatement répartie par deux.

Ne serait-il pas préférable que l’équilibrage se fasse progressivement sur quelques secondes, afin d’éviter de passer brusquement de 0 W sur la batterie 2 à plus de 1000 W d’un coup ?
Si ce phénomène représente bien un risque d’usure prématurée pour les batteries et/ou pour le convertisseur de charge, serait-il envisageable que le correctif (si nécessaire) puisse également être appliqué au moment du basculement d’une batterie à l’autre, lors des phases de charge ou de décharge ?

Il me semble avoir compris, en regardant le code, qu’un facteur de division par 8 est utilisé dans le cas de la charge afin de limiter ces gros à-coups de charge/décharge.

Merci d’avance pour vos éclaircissements.

Cordialement.

---

📧 English version

Hello,

Maybe my question is only based on an assumption, but I would like to ask it anyway.

In the case of two SF2400 AC units:
During charging (and probably also during discharging), if the current exceeds the maximum limit set for one battery, the second battery starts operating. It seems to me that the power is then immediately split in two.

Wouldn’t it be preferable for the balancing to occur gradually over a few seconds, instead of switching abruptly from 0 W on battery 2 to more than 1000 W at once?
If this phenomenon indeed causes premature wear on the batteries and/or the charge converter, would it be possible for the fix (if required) to also be applied at the moment of switching between batteries during charging or discharging phases?

From what I understood in the code, it looks like a division factor of 8 is applied during charging in order to avoid these large charge/discharge spikes.

Thank you in advance for your clarification.

Best regards.

_Originally posted by @Yann33l in https://github.com/Zendure/Zendure-HA/discussions/737_

Youpimatin, 2025-09-09:
Agree with that. Abrupt variations are globally harmful for power electronics.

In the short term (when Zendure will deign shipping the devices I ordered) , I will run three 2400 AC.
Question to @Yann33l : what is your power limit with two 2400 AC (on dedicated sockets) ? 2.4kW, 3kW, 4.8kW ? with or without an electrician app certification document ?

Is the integration limiting the maximum number of Zendure devices that can run simultaneously ?
Can we imagine having four 2400 AC working together to provide 9.6 kW ?

FireSon, 2025-09-09:
@Yann33l That should be fixed in version 1.1.4-pre3. Please read the release notes for that version 😄 
If this version still has problems in this area, please create a new issue!

closed 2025-09-09 by FireSon