Zendure/Zendure-HA #755: [Bug] Local mode: Zendure App Firmware Update -> BMS Firmware page won't loaf

pulsarz, 2025-09-10:
### Home Assistant Version

2025.9.1

### Zendure Integration Version

1.1.4-pre3

### Describe the bug

The page in the Zendure app where you can see the BMS firmware of each individual battery doesn't load when in local mode.

[image]

### To Reproduce

Open the zendure app.
Go to Device settings.
Go to firmware update.
Click on BMS Firmware.

### Expected Behaviour

The page loads and shows the bms versions of the batteries.

### What device are you using?

Hyper2000

### Log Extracts

```shell

```

pulsarz, 2025-09-10:
It is in fact a bug. The other firmware version info does load. Just not this page.

**This is not expected behaviour.**

[image]

harrymayr, 2025-09-10:
The additional BMS Page activly request the info per battery. This is an active communication **to** the device, which is intended to be blocked in local mode.
And there are no other pages which opens and they have the same information (and does not request additional information), as the BMS page: the actual FW.
I still see this not as a bug, but maybe @FireSon has different opinion.