Zendure/Zendure-HA #773: [Feature] Add aggregation sensor for "Output (to Home) energy

deveth0, 2025-09-13:
### Describe the feature

I have a Solarflow 800 with no battery and would like to track the energy provided by the device.

The Zendure App shows the two values "Input from pv" and "Output to home" as Wh (example: I currently have 1.5kWh Input and 0.75kWh to home recorded.

The mqtt integration only shows the first value, which does not help when using the Energy Dashboard.

Could you please add another integration sensor for the power that flows into my home (as provided by the app)? 

(Or do I miss something?)

### What device is this for?

SF800

deveth0, 2025-09-13:
Thanks for pointing that out!
Right now I have a Shelly attached it the device but yeah, a integralsensor makes it way easier!

harrymayr, 2025-09-13:
Hmm, aggr_discharge_total and aggr_charge_total should be the sensors you can use for the Energy dashboard. I use them with my Hyper and I also have no PV panels attached.

[image]

harrymayr, 2025-09-13:
Opps, I oversee "no batterie" ;-). 
Adding this entity is no problem.

harrymayr, 2025-09-14:
> The Zendure App shows the two values "Input from pv" and "Output to home" as Wh (example: I currently have 1.5kWh Input and 0.75kWh to home recorded.

Mmh, w/o batteries, this is a bad efficiency, anyway my PR will solve this, if FireSon implement it.

harrymayr, 2025-09-14:
But in which category of the energy dashboard you want include this? Battery is the wrong thing, and for the PV input the aggr_solar_total is enough.

[image]

deveth0, 2025-09-14:
I'd like to use it for the pv part.
How I understand the values, aggr_solar_total is solar to inverter and that's not necessarily the same value as inverter to home, especially for small panels or shadow when the efficiency goes down (like for me now).

That's most likely also the reason why the app shows two values.